ArduPilot/ardupilot #4028: Reserved space for mission items

lvale, 2016-05-03:
#### Issue details
As discussed with @tridge , Linux based boards don't need to have the restrictions that STM32 based boards have regarding reserved space for mission items

#### Version
All Current versions
#### Platform
- [x] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
NA
#### Hardware type
Linux boards
#### Logs
NA

Locations to revise:

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/AP_HAL_Boards.h#L194

https://github.com/ArduPilot/ardupilot/blob/master/libraries/StorageManager/StorageManager.h#L30

https://github.com/ArduPilot/ardupilot/blob/master/libraries/StorageManager/StorageManager.cpp#L38

https://github.com/ArduPilot/ardupilot/blob/master/libraries/StorageManager/StorageManager.cpp#L64

Perhaps the hw devs of Linux boards might have a word on this, like @vmayoral and @lucasdemarchi.

lucasdemarchi, 2016-05-08:
As talked in person... raising to 128k for Linux boards would be reasonable enough.

lvale, 2016-05-11:
https://github.com/ArduPilot/ardupilot/pull/4082

closed 2016-05-11 by lvale