ArduPilot/ardupilot #4038: ADS-B based vehicle Avoidance

rmackay9, 2016-05-05:
The uAvionix ADS-B sensor is available and capable of providing data on many (50?) near vehicles which we can/should use to avoid vehicles.

Some ideas:

- the failsafe should probably be separated from the ADSB library which should become just a sensor library (like range finder)
- should add pre-arm check that stops vehicle from arming if a possible collision is detected (see algorithm below)
- ideally this would be added after Copter integrates with AP_Arming to reduce duplicate effort but in any case.  This would not allow the vehicle to arm if a sensed vehicle is within the warning altitude, radius and time (see below).

Suggested new parameters:
ADSB_WARN_ALT (i.e. 100m) : altitude above and below our vehicle which, if other vehicle might pass within, will trigger a warning alarm
ADSB_WARN_RADIUS (i.e. 500m) : horizontal distance from our vehicle which, if other vehicle might pass within, will trigger a warning alarm
ADSB_WARN_TIME (i.e. 30sec) : time horizon of warning - i.e. possible breach of our alt or radius must happen within this time
ADSB_FS_ALT (i.e. 50m) : same as ADSB_WARN_ALT except it triggers failsafe behaviour (i.e. dive)
ADSB_FS_RADIUS (i.e. 250m) : same ADSB_WARN_RADIUS except it triggers failsafe behaviour
ADSB_FS_TIME (i.e. 30sec) : same as ADSB_WARN_TIME except it triggers failsafe behaviour
Note: warn-alt, warn-radiu and warn-time are all used together to determine if a warning should be issued

Possible collision detection algorithm:
For our vehicle capture it's current position and (heavily filtered?) velocity
Create a combined velocity vector by adding our vehicle's velocity to the other vehicle's velocity
Create a line segment with it's start at the other vehicle's current position and it's end at start_pos + combined-velocity * warn-time
Calculate the closest point from our vehicle to the line segment (I have a function to do this as part of my terrain following work).
If the distance is within the warn-radius and warn-alt then we have a potential collision.
Potentially filter the output of the above so that the possible collision must persist for a few seconds

How to warn user:
We should create a new MAVLink message which includes:
- Unique ID of the vehicle (ICAO_address should work but may also need callsign and squak)
- callsign (easy for pilot to recognise vehicle)
- projected alt-clearance
- project radius-clearance
- projected time till breach (zero if breach has already occurred)
- state of breach: 0 = not breached, 1 = breached warning area, 2 = breached failsafe area
- recommended action xy (i.e. heading in degrees)
- recommended action z (i.e. climb, descend)
The GCS can use the unique ID and state-of-breach to highlight vehicles (it should have the vehicle position from the ADS-B data passed down to the GCS).  GCS could also call out recommended xy and z actions.

Recommended action could be:
- descend or climb (depending upon whether the vehicle is projected to pass above or below usus)
- fly in a particular direction (i.e. heading)

The Failsafe action should be selectable by a parameter (FS_ADSB_BEHAVE? ADSB_FS_BEHAVE?) which would allow the user to select from a few options:
0 = disable failsafe
1 = only vertical evasive maneuvers (i.e. climb or descend)
2 = only horizontal evasive maneuver (i.e. change heading)
3 = 3d evasive maneuver (i.e. climb/descend + heading change)

As with other failsafes, control would be yanked away from the pilot or autopilotand then the vehicle would perform the action.  Once the threat was cleared the vehicle would RTL.

Concerns:
- ideally the failsafe behavriou should be careful not to hit the ground
- we don't yet support object avoidance so we can't avoid buildings, people.

rmackay9, 2016-05-05:
Here's the useful paparazzi guy's take on this (found by TomP). https://wiki.paparazziuav.org/wiki/MultiUAV#TCAS

jpkh, 2016-05-12:
I think default ranges should be higher. If ADS-B capable airplane is under 250-500m when alarms start go crazy. We chashed already. As currently we are talking commercial jets only. Hopefully in future also small aerial vehicles start to use ADS-B. 
Using time as randy proposed would be a lot better.
I just cannot see ADS-B range breach unless uav is flying next to the airport which should not be possible anyways unless operator have really special permits to do so.
For our use gcs based virual ADS-B system would be a lot better. Or bettwr yet  we could try to look if hw guys can make small uav transponder similar to ADS-B systems that would run on open ISM frequencies.

SamuelDudley, 2016-06-14:
Glider and helis oftern use small ADSB like systems which work on ISM frequency called FLARM. https://flarm.com/products/powerflarm/
My understanding is that the fire fighting service here in Australia had been trialing this tech to integrate small aircraft / UAS into their fire bombing operations.
Might be worth a look.

OXINARF, 2017-07-12:
@rmackay9 Can we close this?

drumadrian, 2017-07-18:
I am currently using a Uavionix PingRX ADS-B receiver on 2 hexacopters.  

One is using a Pixhawk (1st gen) and the other is a Pixhawk 2 based hexacopter with the PingRx plugged into the telem 2 port. 

I can see ADSB Mavlink messages on the 1st gen Pixhawk, but **Not** on the new Pixhawk 2. 

I am using Copter 3.4.6 

I'll try and link to a video of this work.  I'm not sure if this feature is working yet.

magicrub, 2017-07-18:
Check the baud rate? Make sure MAVLink is enabled via protocol=1 or 2

OXINARF, 2017-07-18:
@drumadrian This isn't the place for support questions, use the forum (http://discuss.ardupilot.org) for that. As far as I know Copter 3.5 is what the manufacturer supports in The Cube, not 3.4.

Pedals2Paddles, 2017-07-18:
Make sure your SR0 and SR1 ADSB parameter isn't zero. That's the rate at which ADS-B targets are sent to the GCS. If it's zero, no ADS-B targets are sent.

drumadrian, 2017-07-19:
Hi @magicrub, @OXINARF and @Pedals2Paddles, 

Thank you for your replies.  I'll post more in the forum.  I apologize for the misunderstood message.  I was searching the web for relevant information and I was under the impression that this issue might be closed with a bug.  I'll do my best to study the code base and see where I can offer more insightful feedback in the future.  

Sincerely, 

Adrian

magicrub, 2017-07-19:
This has all been implemented in various forms in the avoidance library. Closing

closed 2017-07-19 by magicrub