ArduPilot/ardupilot #4066: Plane: Bug causes plane to pitch up dramatically

SlipstreamFPV, 2016-05-07:
#### Issue details
Flying my APM Plane today in cruise mode for no apparent reason the plane pitched itself up to 50 degrees. It was definitely a commanded pitch up by the autopilot. The cause isn't apparent to me looking through the logs so I suspect there may be a bug in the TECS code or the stall prevention code. The issue happens just before 12 minutes in the log.

#### Version
APM Plane 3.5.2

#### Platform
Plane

#### Airframe type
My Twin Dream

#### Hardware type
Pixhawk, Dual GPS, Dual Compass, Digital Airspeed, Mauch Current Sensor, MinimOSD

#### Logs
https://www.dropbox.com/s/5vklr2tmpbw3wm1/24.BIN?dl=0

SlipstreamFPV, 2016-05-11:
Anyone had a look at this? It's got me concerned to fly my plane in the automatic modes now...

magicrub, 2016-05-11:
I'll take a peek at the log now

SlipstreamFPV, 2016-05-12:
Awesome. Thanks! Hopefully there is an obvious cause.

magicrub, 2016-05-12:
I don't have any answers for you, but here's a summary of events I see in the log:
- Looks like all the action is just before the 12 minute mark
- RC input 2 briefly spikes from 1550 to 1700 which maps to a user-input of pitch-down.
- Immediately after that the pitch-demand expectantly climbs up from 0 to 35deg which is LIM_PITCH_MAX. The TECS controller's height-demand does not appear to be commanding that.
- RC input 2 (pitch) is moving around a bit, either level or pitch-down.
- Airspeed drops from 15 down to 8 m/s and almost stalls you
- Stall prevention code kicks in: max throttle and pitch-down saves the day but as the airspeed recovers you resube the 35deg pitch up.
- This whole time the RC 2 (pitch) input shows the pilot is pitching down hard.
- Pilot switches to FBWA and RC2 is finally respected and aircraft pitches down and recovers

SlipstreamFPV, 2016-05-12:
Summary matches what happened well. The first pitch down corrections were minor as at the time I had a pan and tilt system on and I was looking around so it was hard to tell exactly what was happening. At first I thought the tilt had jammed up but I realised the plane was pitching once I switched off the pan tilt system. I initially pitched down as you noticed but it didn't hold the downwards pitch so at that point I switched to FBWA to recover with manual ready to go in case it was an issue with the stab system.

Someone suggested that it might have been terrain following kicking in. I had no terrain data loaded at all but I did have terrain following switched on.

magicrub, 2016-05-12:
I thought about the terrain thing but even though you had TERRAIN_ENABLE=1, you had TERRAIN_FOLLOW=0 so your autopilot is only configured to fetch the data but not actually use it.

magicrub, 2016-05-12:
And also, the terrain log data shows the height data as all zero so that was definitely not the problem.

SlipstreamFPV, 2016-05-13:
What about the GPS 3D velocity estimations, could they have come into play? I have dual GPS units but I did notice the sat count changes right before the glitch.

tridge, 2016-05-13:
Hi @SlipstreamFPV and Tom,
Peter and I have also had a look at the log. We don't yet know what happened, but it does look like something in TECS. The demanded pitch from TECS matches CTUN.NavPitch.
Do you happen to have the log of this flight as well? It's possible we are missing some critical log messages in the DF log and perhaps the tlog could offer some clues.
Cheers, Tridge

SlipstreamFPV, 2016-05-13:
The tlog is only recorded if a data link is used correct? If that's the case I don't have one unfortunately. The only other record of the flight I have is the DVR footage which shows the event but doesn't obviously contain any more detail.

tridge, 2016-05-13:
I have created a PR to address this issue here:
https://github.com/ArduPilot/ardupilot/pull/4119
the PR constrains the specific energy integrator to not change too fast. In the log the iph value changes suddenly by its full range. It is now limited to changing not more than 10% of full range per cycle, which will prevent glitches from causing this sort of pitch spike.
I also added logging of the internal energy variables which must be behind the problem, so if we get glitches in the future we'll have a better idea of what causes them
Tom, can you have a look at the PR ?

WickedShell, 2016-05-13:
I think it's a separate issue but semi related. I was in guided at 300m set
the target altitude to 120m. The aircraft hit 27m/s airspeed trying to
descend faster. Since this exceeds our maximum of 22m/s by a good margin
and it held it for a long time I have to think tecs wanted it. I haven't
gotten to review that log or the changes yet, I just wanted to mention it
as a place where tecs forced the aircraft past limits with a large risk to
flight safety.

I will hopefully look at these changes and that log within a couple of
days, but don't wait on then for me to look at them.

tridge, 2016-05-13:
@WickedShell, can you link to the log of that overspeed problem?

magicrub, 2016-05-13:
@tridge I posted some comments to PR https://github.com/ArduPilot/ardupilot/pull/4119

WickedShell, 2016-05-16:
@tridge the log is here: http://files.swiftradioplanes.com/30.BIN the overspeed occurred while in a guided descent. I haven't looked through it yet in detail. (There is also a serious looking EKF panic (that it recovered from) part way through the mapping mission that I'm hoping to get Paul to look at).

magicrub, 2016-06-22:
closed, included in Plane release v3.6.0

closed 2016-06-22 by magicrub