ArduPilot/ardupilot #4069: Follow me mode does not work

kudykam, 2016-05-09:
#### Issue details
When I switch to follow me mode, quad do nothing for 10 sec then start descending (and crash).

#### Version
3.4 dev (May)

#### Platform
- [ ] All
- [ ] AntennaTracker
- [X ] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk 2.4.8

#### Logs
https://www.dropbox.com/s/7diaagwrgogq51y/bin-logs.zip?dl=0
(the latest one)

magicrub, 2016-05-09:
What GCS were you using?

rmackay9, 2016-05-09:
Txs for the report.  Of course Copter-3.4-dev is "bleeding edge" software.  It hasn't even begun beta testing yet and as such is "use at your own risk".

kudykam, 2016-05-09:
Tower app on mobile.
Tried v 3.3.3 before, but copter fly terrible and had EKF compass error. On 3.4 its fine.

kudykam, 2016-05-27:
I tried followme today with reinstalled system on my android phone Xiaomi mi4c with miui 7. Now quad hoover on one spot and do nothing when switched to followme. Doesnt matter if I change radius, altitude, or follow me mode.

lvale, 2016-05-27:
There will be a new method of follow-me, now that Mavlink already has a specific message to handle that situation. 
Also, GCS's like Tower will have to implement that also

https://github.com/ArduPilot/ardupilot/issues/3929

kudykam, 2016-05-28:
I tested follow me again today, make sure my phone has gps lock. When in loiters simple mode, I swithce to follwo me, then set high to 3m (was in 2) then quad start fast descending and crash.

When can we expect another dev release of new firmware?

(off topic: Can I set descent speed in LOITER mode nad ALT HOLD mode separately? pilot_velz_max works for both)

OXINARF, 2016-05-28:
> When can we expect another dev release of new firmware?
Beta testing of 3.4 starts today! You can see and discuss it at http://discuss.ardupilot.org/t/copter-3-4-beta-testing-start/9290

rmackay9, 2016-05-28:
@kudykam, I'll give follow-me a try next week (hopefully early next week) and see if I can reproduce the issue.  I've had a quick look at the logs and the vehicle is trying to descend.  Maybe the changes to support absolute altitudes mean that we are now interpreting the Tower messages differently (i.e. before we assumed the target altitudes were all alt-above-home when Tower may actually have been sending absolute altitudes).  Anyway, txs for the report - we should get to it soon.

lvale, 2016-05-28:
@rmackay9 We don't have implemented yet https://github.com/mavlink/mavlink/issues/509

Then it will be a matter of the GCS's to also support it

rmackay9, 2016-05-30:
@lvale, but just to be clear, Tower uses constant position updates to implement follow-me in the Tablet.  It's not a great way to do it but that's how it's done and it should work.

rmackay9, 2016-06-28:
I've confirmed the issue is with the MAV_FRAME being sent by tower so I've created an issue here for Tower: https://github.com/DroidPlanner/Tower/issues/1761.
This issue doesn't occur with Mission Planner.  It's a serious issue of course but I'm closing it here because it's not technically a flight code issue.

closed 2016-06-28 by rmackay9