ArduPilot/ardupilot #4071: Auto Take Off (Hand Launch) Abnormal Roll

onurboy, 2016-05-09:
#### Issue details
In plane wiki page, it's stated that we need to point the aircraft to the take off direction before switching the mode to auto because it records the direction while changing the mode before take off. Is it really needed and critical or is it just a notice?
What would plane do if we switch from FBWA to Auto while plane is on the ground and nose of the plane is directed to another point from the direction that it is really going to take off?
I am asking this issue because sometimes I have abnormal roll action during auto take off (hand launching) and I have to compansate roll to other side to be able to save the plane. I've noticed that there is a statement in wiki as I mentioned above? I did not really care this statement before. Do you think it is the cause?

Thank you all.

Onur.

#### Version
_3.5.0, 3.5.3_

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [x] Plane
- [ ] Rover

#### Airframe type
Skywalker 2013 1880 wings T-Tail

#### Hardware type
Pixhawk

#### Logs
https://www.dropbox.com/s/m4nusidqfbpkl93/1%208.5.2016%2010-21-32.bin?dl=0

magicrub, 2016-05-09:
It is no longer needed to be pointed in the launch direction when switching
to auto for a takeoff, the wiki needs to be updated. Your roll must be
something else. Is it just roll or does it actually navigate to somewhere
else?

WickedShell, 2016-05-09:
Tom if the safety button hasn't been pressed or you have a acceleration
delay then that statement is true, but if you don't have any delay, and the
aircraft can spin the motor it will use the wrong heading.

onurboy, 2016-05-09:
@magicrub,
Actually I think it wants to turn to somewhere different from the taking off direction, thats why it makes a hard roll to right or left. Since it's in the take off phase, the speed is relatively low and plane is trying to climbing, that abnormal roll or turning movement reduces the lift of wings, nose goes down and I need to give eleron to the opposite side and back elevator command from the radio in order to make plane level and climbing. After my correction it climbs and goes to the first waypoint flawlessly and even auto landing with no problem.

@WickedShell,
I have an acceleration delay, I mean motor starts after I throw the plane into the air.
But I dont know why my plane does something like that even it's rare.
The log file I have attached above is one of the problematic auto take off that I needed to correct the plane during auto take off.
Thank you.

onurboy, 2016-05-09:
@WickedShell,
What do you mean by saying " if the safety button hasn't been pressed "?

WickedShell, 2016-05-09:
@onurboy if you are using the safety switch with pixhawk. When the safety is disarmed (blink LED, io processor reported as safe). which means that the outputs on channel1-8 are inhibited, the plane will not lock heading. It's something I use a lot, but the majority of plane users probably do not use in this manner.

onurboy, 2016-05-09:
@WickedShell ,
I use safety switch. My action order in the field is;
1. Turn on TX,
2. Connect batt to Airplane,
3. Connect to airplane from mission planner,
4. After GPS lock push safety switch,
5. Arm airplane,
6. Switch mode to auto

All defined above done when the plane on the ground and stationary with a random direction.
Then I grab the plane, remove the pitot tube cover, consider the wind and turn into the wind then throw the plane.

Do you think that this procedure cause the problem that I mentioned above?

moreba1, 2017-09-21:
i also have this problem. in hand launching skywalker x8 heading change before to reach takeoff altitude.

tomipiriyev, 2017-10-21:
Hi I am getting the same issue, airplane turns before takeoff altitude, makes very dangerous turns.

Have you solved this problem? I am using Arduplane 3.8.2

moreba1, 2017-10-27:
can anyone solve this problem?

WickedShell, 2017-10-27:
Typically this is either associated with:
A) A GPS glitch around when the GPS based ground course is read and locked for the takeoff, causing the system to lock a desired heading that is off from the direction the aircraft was launched
B) The aircraft attempts to maintain the heading the direction it was thrown, but it does not attempt to maintain a ground track that direction. What this means is any crosswind component during takeoff will push the aircraft sideways over the ground. (This behavior definitely needs to change)

moreba1, 2017-10-27:
thanks,
i dont think this problem is because of GPS glitch or compass erros . the takeoff heading is very different from takeoff throw (about 60 degree). also in my case there was no wind at that day.

WickedShell, 2017-10-27:
@moreba1 Logs are needed or this a speculation game with no clear anwsers :)

moreba1, 2017-10-27:
yes . i attached link of telemetry log.
http://s9.picofile.com/file/8310217126/2017_08_24_10_15_07.tlog.html

tomipiriyev, 2017-10-27:
Please share your last 3 flight log .   I found the reason of this problem was from servos.  With remote controller when i move no problem at all.  But without power. I when i move with my hand , there was some differences , one of them was a little bit harder.   I changed servos and problem solved.  But yout problem might be different need to loom at your log . And does airplane rols to same direction every time?

moreba1, 2017-10-28:
@WickedShell 
could you find the cause of problem from tlog ?

WickedShell, 2017-10-31:
@moreba1 I can't figure out what the download button is on the first link, and on your second log whatever tool you used to convert it messed up badly. A lot of messages are dropped from the log file, and many of the fields on an actual message are dropped (IE if a message contained 5 fields only 2 are in the converted file)

moreba1, 2017-10-31:
http://s9.picofile.com/d/8310217126/30ed19e3-053b-4064-bd8f-11aecd87c221/2017_08_24_10_15_07.tlog
i think my problem is from tip stall in takeoff then uav rotate .

WickedShell, 2018-08-26:
Closing, this is a support issue, and looks to have been solved in the end. For more support please go to discuss.ardupilot.org.

closed 2018-08-26 by WickedShell