ArduPilot/ardupilot #4072: Flying wing differential spoilers AND crow flaps

raptortrias, 2016-05-09:
#### Issue details
For flying wing with 4 separate control surfaces (2 pair wing) it would be convenient to be able to setup both differential spoilers for YAW as well as crow flapss for AUTO_LANDINGS

#### Version
NA

#### Platform
- [ ] Plane

#### Airframe type
FLYING WING

#### Hardware type
PIXHAWK compatible
#### Logs
NA

magicrub, 2016-05-09:
depends on https://github.com/ArduPilot/ardupilot/pull/2935

jgilbertfpv, 2018-09-01:
Could this not be added to the servo.cpp file?
if (flap > 0) {
// apply flap to right wing
dspoiler1_right = constrain_float(elevon_right + flap, -4500, 4500);
dspoiler2_right = constrain_float(elevon_right - flap, -4500, 4500);
// apply flap to left wing
dspoiler1_left = constrain_float(elevon_left + flap, -4500, 4500);
dspoiler2_left = constrain_float(elevon_left - flap, -4500, 4500);
}

Naterater, 2018-12-28:
Complete. See https://github.com/ArduPilot/ardupilot/pull/9391

closed 2018-12-28 by Naterater