ArduPilot/ardupilot #4075: ROS/Gazebo Simulator unable to start

glody, 2016-05-10:
#### Issue details
_Setup "ROS/Gazebo Simulator" according wiki "http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html" and try to start the simulator by command "sim_vehicle.sh -f arducopter_sitl_ros --console", Gazebo windows doesn't show, message "[launcher.launch] is neither a launch file in package [arducopter_sitl_ros] nor is [arducopter_sitl_ros] a launch file name
The traceback for the exception was written to the log file" shows. Please check logs below_

#### Version
_NA_

#### Platform

- [ ] Copter

#### Airframe type
_ROS/Gazebo Simulator with SITL_

#### Hardware type
_NA_

#### Logs

[start_gazebo.txt](https://github.com/ArduPilot/ardupilot/files/256509/start_gazebo.txt)

[gazebo_output.txt](https://github.com/ArduPilot/ardupilot/files/256510/gazebo_output.txt)

squilter, 2016-05-11:
It works fine for me. Are you sure that you didn't miss a step in the directions?

glody, 2016-05-19:
I missed to add folder 'catkin_ws/src' to $ROS_PACKAGE_PATH. which is the root cause of the error above.

After fix that, I still unable to start gazebo, I tried many times, and didn't find miss steps. Could you find the reason from the log below? Thanks.
[gazebo start error.txt](https://github.com/ArduPilot/ardupilot/files/271998/gazebo.start.error.txt)

khancyr, 2016-05-19:
hi,

You got : ERROR: cannot launch node of type [mavros/mavros_node]: can't locate node [mavros_node] in package [mavros]
ERROR: cannot launch node of type [joy/joy_node]: joy
Did you install mavros and joy package ? Did you source you catkin workspace ? (in the root of you catkin workspace use : source devel/setup.bash)

squilter, 2016-10-29:
The wiki instructions used are now outdated, and will be overhauled soon.

closed 2016-10-29 by squilter