ArduPilot/ardupilot #4076: Segmentation fault ArduCopter-quad 3.4 dev navio+ / raspberry PI2 / raspian jessie kernel 4.4.x

philippelecoq, 2016-05-10:
#### Issue details
Segmentation fault
ArduCopter-quad 3.4 dev navio+ / raspberry PI2 / raspian jessie kernel 4.4.x
from daily build here : http://firmware.ardupilot.org/Copter/
- ArduCopter-quad-navio-2016-04-20.elf ==> OK
- ArduCopter-quad-navio-2016-04-30.elf ==> Segmentation fault

#### Version
ArduCopter-quad 3.4 dev 

#### Platform
- [ ] All
- [ ] AntennaTracker
- [X] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
Quad

#### Hardware type
emlid Navio+ and Raspberry PI2 + raspian jessie kernel Linux navio 4.4.6-emlid-v7 #1 SMP PREEMPT 

#### Logs
strace : ArduCopter-quad-navio-2016-04-30.elf :
rt_sigaction(SIGRTMIN, {0x76d1e434, [], SA_RESTORER|SA_SIGINFO, 0x76c061a0}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {0x76d1e2d8, [], SA_RESTORER|SA_RESTART|SA_SIGINFO, 0x76c061a0}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
getrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM_INFINITY}) = 0
clock_gettime(CLOCK_MONOTONIC, {12126, 359253031}) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0} ---
+++ killed by SIGSEGV +++
Segmentation fault

lucasdemarchi, 2016-05-14:
This should be fixed now on master.

lucasdemarchi, 2016-05-14:
If it still crashes, feel free to reopen.

closed 2016-05-14 by lucasdemarchi