ArduPilot/ardupilot #4080: How to import and build the project in eclipse

SalahuddinAsh, 2016-05-10:
#### Issue details
Hi guys
I'm a newbie with this project and with importing projects from github and working on them in eclipse
I don't want to know how to branch, clone or join
I just need to know how to import the project in eclipse and build it.
I tried searching for some tutorial but it didn't work .
In eclipse when it comes to **"Import existing Project**, it says "no projects found"

I would be so grateful if someone helped me

thanks

#### Version
_What version was the issue encountered with_

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-05-10:
Here you go: http://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html#editing-the-code-with-eclipse

closed 2016-05-10 by magicrub

SalahuddinAsh, 2016-05-12:
Hi, thanks a lot for the link
I tried to follow the installation instruction but a problem appeared
After installing eclipse, the Make tab on the right has the buttons "New Make target" and "Build make target" inactivated!!
And I don't know where the problem is?
I'm working on windows 7
Any suggestion please?
Thanks

jpkh, 2016-05-12:
Github issue lists are more or less for the bug reports, merge requests etc. Technical / problemsolving issues are best to post at http://discuss.ardupilot.org forums
Please post there, your questions will be handled there easier and better.

magicrub, 2016-05-12:
As @jpkh said, this is a bug tracking tool not a help tool. Please visit the forums at discuss.ardupilot.org. This issue is closed.

SalahuddinAsh, 2016-05-12:
I'm sorry guys for asking in the wrong place, I just didn't know it was the wrong place.

THanks