ArduPilot/ardupilot #4088: Calibration not working in sitl

khancyr, 2016-05-11:
#### Issue details
Hi,
@guludo 

The calibration module doesn't work
It blocks on "sitl_accelcal: %s - Waiting for attitude"
It seems that attitude change are not send to SIM. (see in SIMSTATE mavlink message)

#### Version
master with sim_vehicule.py

#### Platform
- [X] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
All

#### Hardware type
Sitl

guludo, 2016-05-11:
Hi @khancyr

You need to run the sitl binary with the "calibration" model. That's isn't on sim_vehicle.py yet. It's a good idea putting it there. In the meanwhile, you can just run you binary with `--model calibration `. Unfortunately, you can't calibrate and fly with the same model... Maybe we could add that feature in the future if that's needed.

khancyr, 2016-05-11:
Thanks for the explanation. Could you add it to calibration Readme ? 

So I got a different problem not related to calibration

closed 2016-05-11 by khancyr