ArduPilot/ardupilot #4090: APMrover 2 Sitl not arming

khancyr, 2016-05-11:
#### Issue details
It is currently impossible to arm rover in sitl with sim_vehicule.py
Here is mavproxy output for arm throttle
````
Init ArduRover v3.0.0 (b7d1e047)

Free RAM: 131072
Firmware change: erasing EEPROM...
done.
load_all took 0us
0 0 0 DataFlash_File: buffer size=16384
online system 1
Mode INITIALISING
fence breach
APM: ArduRover v3.0.0 (b7d1e047)
APM: Barometer calibration complete
APM: <startup_ground> Ground start
APM: Warming up ADC
APM: Beginning INS calibration. Do not move vehicle
APM: Initialising APM
Init Gyro**
APM: Ready to drive
APM: GPS 0: detected as u-blox at 38400 baud
GPS lock at 0 meters
Mode MANUAL
APM: Init HOME
height -1170
height 0
EKF2 IMU0 tilt alignment complete
EKF2 IMU0 yaw alignment complete
EKF2 IMU1 tilt alignment complete
EKF2 IMU1 yaw alignment complete
EKF2 IMU0 Origin Set
EKF2 IMU1 Origin Set
EKF2 IMU0 is using GPS
EKF2 IMU1 is using GPS
APM: PreArm: 3D accelerometers calibration needed
APM: PreArm: GPS 1 failing configuration checks
APM: GPS 1: u-blox navigation rate configuration 0x1FFF
Got MAVLink msg: COMMAND_ACK {command : 400, result : 4}
````

It asked for acc Calibration. And got a problem with GPS...
This problem is not present in copter

Disable ARMING_CHECK works 

#### Version
master with sim_vehicule.py

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [X] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
Sitl

#### Logs
_Please provide a link to any relevant logs that show the issue_

gmorph, 2016-05-12:
I'm unable to re-produce this issue.  Have a good read through this wiki page on running SITL.
http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html
Note that initial load from the parameters file that you MUST do we as we need small INS_ACC offsets so INS is recognised as being calibrated.
You might also need to check how your GPS is configured.
Alternatively you can start from scratch with the "sim_vehcile.py -w" which will wipe everything in the simulator and start again although you will need to load the parameters file described in the wiki page linked above.
I'm going to close this as I don't think its a bug in the system.
Thanks, Grant.

khancyr, 2016-05-12:
To reproduce : 
sim_vehicle.py -v APMrover2 --map --console 
Then when init done : just arm throttle and you got the error

Using -w change nothing since it is already default param

Loading /Tools/autotest/Rover.param fixe arming. But we still got GPS config error
````
APM: PreArm: GPS 1 failing configuration checks
APM: GPS 1: u-blox navigation rate configuration 0x1FFF
APM: Throttle armed
````

I am reporting the issue since there is a divergence with copter sitl

gmorph, 2016-05-12:
That's the key - to start it with the -v from the ardupilot folder and I get the same problem.  However, if I cd to the APMrover2 folder and run it from there without the -v everything is fine.  I've re-opened.
Peter, one for you :-)
Thanks, Grant.

gmorph, 2016-05-12:
Note I've just tested with sim_vehicle.sh and it has exactly the same problem.

peterbarker, 2016-05-12:
@gmorph  Investigating.  Try starting it in the APMrover directory *after* moving aside your current eeprom.bin - I'm betting it won't arm for the same reason.

I wonder if the default parameter set is insufficient to let this arm.

If you could do a mavparmdiff between a working eeprom.bin and a non-working eeprom.bin that would be handy.  I *didn't* move my working eeprom.bin aside ;-)

khancyr, 2016-05-12:
Just test inside or outside APMrover2 directory, with sim_vehicule.py alone or with -v option, with and without loading param. ( cleaning each time with git clean)

Directory and -v option haven't any influence.
Loading param (Tools/autotest/Rover.param) solve pre-arm accel issue

GPS issue is still present
According to code, it is because at pre-arm check, it cannot retrieve navigation rate parameter from GPS.

khancyr, 2016-05-12:
I fixe loading default parameters like in copter and plane.
But I didn"t find a clue about GPS bug

peterbarker, 2016-05-12:
@khancyr My SITL rover arms after applying your patch.  No GPS fault.

magicrub, 2016-05-13:
is this ready to be closed?

gmorph, 2016-05-14:
Fixed.

closed 2016-05-14 by gmorph

AndWik, 2016-07-18:
I have the above issue with Plane SW but it's a common source I guess. With a CUAV M8N GPS the problem is:
To have the GPS working at all I have to enable GPS-Config but I cannot arm unless I disable it after power up. Otherwise I get the error GPS1 configuration failure.

Anders
Sweden