ArduPilot/ardupilot #4128: Error compiling Please dont tell compile master same problem

RaviPrasadWA, 2016-05-14:
#### Issue details
error: unrecognized command line option "-std=gnu++11"

#### Version
Copter-3.3

#### Platform
- [ ] All
- [ ] AntennaTracker
- [*] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
ArdupilotMega atmega2560

#### Logs
_Please provide a link to any relevant logs that show the issue_

RaviPrasadWA, 2016-05-14:
OS - Ubuntu12.04
GCC - 4.6.3
G++ - 4.6.3
AVR-GCC - 4.5.3
AVR-G++ - 4.5.3

Command to compile  : make apm2

magicrub, 2016-05-15:
APMv2 hardware support was dropped before v3.3. Ran out of memory.

closed 2016-05-15 by magicrub

kwikius, 2016-05-15:
Your compiler is out of date. You need AVR-GCC - 4.7 or higher.  That is the first version to support -std=gnu++11