ArduPilot/ardupilot #4129: Not able to upload the Arducopter-3.2.1 firmware

RaviPrasadWA, 2016-05-15:
#### Issue details
ERROR: address 0x40010 out of range at line 16389 of /tmp/ArduCopter.build/ArduCopter.hex

#### Version
ArduCopter-3.2.1

#### Platform
- [ ] All
- [ ] AntennaTracker
- [*] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
ArduPilotMega atmega2560

#### Logs
/usr/bin/avrdude -c stk500v2 -p atmega2560 -P /dev/ttyACM1 -b115200  -U flash:w:/tmp/ArduCopter.build/ArduCopter.hex:i -D

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801
avrdude: reading input file "/tmp/ArduCopter.build/ArduCopter.hex"
avrdude: ERROR: address 0x40010 out of range at line 16389 of /tmp/ArduCopter.build/ArduCopter.hex
avrdude: read from file '/tmp/ArduCopter.build/ArduCopter.hex' failed

avrdude: safemode: Fuses OK

avrdude done.  Thank you.

RaviPrasadWA, 2016-05-15:
OS - Ubuntu12.04
GCC - 4.6.3
G++ - 4.6.3
AVR-GCC - 4.5.3
AVR-G++ - 4.5.3

Command to compile : make apm2
Command to upload : make upload

RaviPrasadWA, 2016-05-15:
And yes the port is /dev/ttyACM1 not /dev/ttyACM0

magicrub, 2016-05-15:
Does anyone know if v3.2.1 supported apmv2 hardware? I thought it did but
this compile error says the binary is too big which is why we stopped
supporting it.

Ravi, did you change the code any or are you compiling the code exactly on
the v3.2.1 tag?

kwikius, 2016-05-15:
He could try a later version of avr-gcc. Later versions should produce more compact code. 
Try avr-gcc4.7.x

rmackay9, 2016-05-16:
Copter-3.2.1 was the last version to work with the APM 2.x boards.  The issue is that it barely fits so, as @kwikius says, a different compiler may help.  Alternatively he/she may be able to make it fit by disabling some features in APM_Config.h.

closed 2016-05-16 by rmackay9

RaviPrasadWA, 2016-05-17:
Hi guys APM_Config.h is not at all modified even the code is not modified guys I am approaching on a solution to make cheaper/simpler OPen source solution out of the ardupilot henceforth the use of 2560mega much cheaper than pixhawk any how Will try it with avr-gcc 4.7.x or if any of you could help me reduce the code I would grateful for that also ...... AND THANKS FOR THE SWIFT REPLIES ...... HOPE TO MAKE ARDUPILOT THE NEXT BIG THING

RaviPrasadWA, 2016-05-17:
WOOOOOOOOOOOOOOOW I just disabled a few features in APM_Config.h and added a -D flag in the avrdude command its ALIVE .................. ( WORKING ..... )

magicrub, 2016-05-17:
Congrats!