ArduPilot/ardupilot #4130: Need ratio for different throttle output fot coaxial copter

lokanaft, 2016-05-15:
#### Platform
- [x] Copter

#### Airframe type
Coaxial

I test two coaxial motors on thrust stand with different throttle level and found with same thrust 500gr:

Throttle level _M1 / M2_ | Current
------------ | -------------
50% / 50% | 2A
47% / 52% | 1.85A

#### How I see to use it:

1. Set "Ch6 Opt" for this ratio function parameter
2. In flight "Loiter" try different ratio with Ch6
3. After, open log and find minimal current, find for this moment Ch6 level and set value for this parameter forever

lthall, 2018-01-12:
Duplicate #2479

lokanaft, 2018-01-12:
No, not duplicate, absolutely different

IamPete1, 2021-07-25:
This is possible in flight via the new dynamic scripting motors back-end, happy to help with the lua script if anyone wants to give it a try. In the mean time I will close this.

closed 2021-07-25 by IamPete1