ArduPilot/ardupilot #4135: Optical/ESC Rotor RPM Sensor support and feature request

Jman841, 2016-05-16:
details
The majority of major ESC's now have an output for RPM sensor, or, the use of an Optical sensor to measure the rotor speed on Arducopter Traditional heli can be used for a governor and as a auto-rotation fail-safe. Currently, if in an auto mode and you have a motor failure the helicopter will pitch up until it looses all rotor speed and falls out of the sky. If we have the RPM measurements, preferably an optical sensor as it will not rely on the ESC, if there is a power failure in flight, the autopilot can reduce the pitch to maintain a minimum required rotor RPM to bring the heli down softer than a complete fall. The ultimate goal would be to integrate a final flare stage using a laser sensor as well.

#### Version
Arducopter Traditional heli 3.3.3

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ X] Copter
- [ ] Plane
- [ ] Rover

#### Airframe type
Traditional Heli

#### Hardware type
All

R-Lefebvre, 2016-05-17:
RPM input is already supported in the code.  I'm not sure if this is detailed in the wiki.

I've also got some governor code working already but it's not in Master.

kozinalexey, 2016-08-20:
hi , i found rpm pwm code present in code, but not found how to configure input pin for pwm rpm sensor, help me please

Georacer, 2017-02-03:
Perhaps this link can prove useful:
http://discuss.ardupilot.org/t/how-to-read-rpm-signal-from-esc/8326/23

palm369, 2018-08-01:
I would like to push this topic.
It would be very useful to detect a motor failure over the RPM input to give alerts.
Right now on a plane when your motor fails for any reason and there is not thrust, the plane will not notice it. If we compare throttle output to actual RPM, we can detect it and print out a error.

IamPete1, 2021-08-15:
I'm going to close this, we have RPM sensing and autorotation in the trad heli code now.

https://ardupilot.org/copter/docs/traditional-helicopter-autorotation-mode.html?

 There is a lest one issue dedicated to detecting motor failure on plane. https://github.com/ArduPilot/ardupilot/issues/1861

closed 2021-08-15 by IamPete1