ArduPilot/ardupilot #4136: Booting issue with PX4Mini AIO board on all firmware versionq

rscarawa, 2016-05-16:
#### Issue details
I have a new Mini AIO board that initially worked and now seems to fail on boot.  I have loaded many different firmwares on to this board but now only get a red led.  I have never flown with this board as I am still configuring the plane which this will go in.  The messages I get are:

From BOOT file:  No PX4IO board found failed to start mtd driver 
From px4io_update:  check Safety Button check Safety Button 
From the other px4io_update:  Loading /etc/px4io/px4io.bin Failed loading /etc/px4io/px4io.bin 

#### Version
PX4MINIAIO

#### Platform
All platforms.
#### Airframe type
Flying wing

#### Hardware type
PX4 MINI V1.3 FLIGHT CONTROLLER

#### Logs

Not much in the logs.  I posted everything above that is in them.  The board seems to accept new firmware.  Just does not boot.  Nothing special happened between the last time it worked and when it failed.  I get the impression that I need a file on the card to make this work or the new firmware does not actually erase everything in memory when it does its upgrade.  Any help would be appreciated.

rscarawa, 2016-05-16:
I have also tried formatting the card and tried different sd cards with no success.

magicrub, 2016-05-16:
I don't know much about the px4io but that error message sounds like a
hardware failure to me.

rscarawa, 2016-05-16:
Looking inside the terminal window, I get this error when trying to start MTD.

nsh>  mtd start
mtd:  failed to initialize EEPROM driver.

Is that a hardware issue?

Naterater, 2019-01-07:
Can this be reproduced on other hardware? If not, I don't think this will be solved and we should probably close this issue.

WickedShell, 2019-04-05:
Closing as this is a support issue, and future development on PX4 based firmware has ended. (It's also a stale issue). If you need more help please go to discuss.ardupilot.org.

closed 2019-04-05 by WickedShell