ArduPilot/ardupilot #4140: Failed to build Examples

glody, 2016-05-17:
#### Issue details
_
"collect2: error: ld returned 1 exit status" while running test "./Tools/autotest/autotest.py build.Examples".

Git bisect result:
commit 7814841cd6c2222bdb5bd86ffc5b132c873b1bf2
Author: hiro2233 <htakey@gmail.com>
Date: Fri Apr 1 05:03:22 2016 -0400
AP_HAL: Add RCOutput interactive example with Menu
:040000 040000 b1d241a7e99a934c5993bd748d707d76b711b7d0 d7b8e09d7833f75d154d2d1b279d50564184e9c5 M libraries
_

#### Version
_master branch:e3b2e90a2762684f240e481a285b46f6de0500ab_

#### Platform
- [ ] Copter

#### Airframe type
_NA_

#### Hardware type
_SITL_

#### Logs
_
[build.Examples.txt](https://github.com/ArduPilot/ardupilot/files/267462/build.Examples.txt)
_

lucasdemarchi, 2016-05-17:
make build system is deprecated for Linux  boards. From your log I see it fails for you with the navio board. I just checked with waf and it works.

What we probably need to do is to change the build_examples.sh script to use waf instead of make.

peterbarker, 2017-12-19:
Examples are now built with waf.

closed 2017-12-19 by peterbarker