ArduPilot/ardupilot #4144: SITL mission no longer working on current master.

mday299, 2016-05-17:
#### Issue details
my Plane won't take off anymore (running off of ArduPlane/master). Looks like the set of commits that changed "is_zero" may be the culprit. If I'm on this commit:
ArduPilot/ardupilot@48c243b

takeoff is broken, but if I'm on this one:

ArduPilot/ardupilot@ebd993d

all is well.

#### Version
Current master.  See also notes on specific commits in "Issue details" above.

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [X] Plane
- [ ] Rover

#### Airframe type
Simulated Rascal. :)

#### Hardware type
Lubuntu 14.04 running on Core i7 laptop.

#### Logs
Attaching logs and mission.

[logs.zip](https://github.com/ArduPilot/ardupilot/files/268229/logs.zip)

magicrub, 2016-05-17:
I can confirm this. The template can not handle AP_Float so a temporary workaround is to change is_zero(var) to is_zero(var.get()) but really is_zero needs to be fixed ASAP.

OXINARF, 2016-05-17:
@magicrub According to what was said before, that's not a workaround but the correct way to write it. I'm not very familiar with this, so I'm tagging @lucasdemarchi and @tridge here.

OXINARF, 2016-05-17:
I was looking at the template and it seems that it was supposed to handle AP_Float, I would say the issue is with std::numeric_limits<T>::epsilon() but I'm not sure. Tagging @dgrat too.

lucasdemarchi, 2016-05-17:
I thought sitltest would catch this problem, sorry for the trouble.

We are calling fabsf() so there's no point in trying to figure out a different epsilon. It's now fixed on master.

lucasdemarchi, 2016-05-17:
Actually let me push to a PR and close this when it's applied:

magicrub, 2016-05-17:
Thanks Lucas

magicrub, 2016-05-17:
Michael, I'm out at the field right now, are you able to test out Lucas's
patch? If not I'll get to it tonight.

lucasdemarchi, 2016-05-17:
I actually tested a takeoff in sitl. Just wanted someone to confirm the
fix. I can push it.

magicrub, 2016-05-17:
But does it compile in cygwin on make?

OXINARF, 2016-05-17:
Someone should write a SITL test for this, because it didn't fail on Travis (but we only test QuadPlane there) neither in our Autotest server.

dgrat, 2016-05-17:
@lucasdemarchi 
just for interest, what's wrong with std::numeric_limits<T>::epsilon()

mday299, 2016-05-18:
@magicrub Tom: appears fixed for Plane as far as my tests show.

khancyr, 2016-05-18:
@dgrat just see in example that std::numeric_limits<T>::epsilon() must be scaled to the magnitude of values used but we don't do it
http://en.cppreference.com/w/cpp/types/numeric_limits/epsilon

dgrat, 2016-05-18:
@khancyr 
No this is not true. It is a template function returning FLT_EPSILON for a float a parametet.

dgrat, 2016-05-18:
The fix is essentially a very bad one, because it is not accurate for doubles or anything else. Anyway, I am out of this.

lucasdemarchi, 2016-05-18:
There was 2 problems. First is that we were calling fabsf() that returns a float and then trying to compare to epsilon from the type. This is wrong... You'd need to at least use std::abs(). The second is that if T is AP_Float the epsilon is should be from float not from AP_Float.

lucasdemarchi, 2016-05-18:
@dgrat I know the fix is not optimal. But leaving the master branch broken is worse.

RoverHunter, 2016-07-15:
I am having the same problem with f579c3e6ebc3531e8fa778cba44eefe02db13dec. I'm using Mission Planner 1.3.39.1 build 1.1.6040.17593 and XPlane 10.45.
Other thing I have noticed is that under Simulation tab, I have no data under Plane GPS, Plane IMU, Ardupilot Output and Autopilot Status.
Apparently Xplane is communicating with Mission Planner, but the plane is unable to takeoff.
Shouldn't e0eecd56b2a329775d9fc731ce047a9ce1228558 resolve the AP_Float problem?

lucasdemarchi, 2016-07-15:
@RoverHunter works fine for me. Could you open another issue, it's likely a different problem... and make sure to tell what exactly doesn't work.

I tested here with master branch with ArduPlane in sitl + mavproxy and it works fine with Tools/autotest/CMAC-circuit.txt mission.

Thanks.

closed 2016-07-15 by lucasdemarchi