ArduPilot/ardupilot #4153: Question - Source for SET_POSITION_TARGET_LOCAL_NED

alexblack, 2016-05-18:
Hi, I'm trying to get my 3DR Solo drone to make precise movements (without GPS) by sending the `SET_POSITION_TARGET_LOCAL_NED` message.

Where can I find the source for this so I can get a better sense of how its implemented? I am not familiar with the ArduPilot source yet.

Thanks!

rmackay9, 2016-05-18:
I'm afraid the EKF only knows it's position and velocity if it has a GPS attached (or an optical flow).  It's not possible for it to know these values just from the IMU.

As a side note, we get this question pretty much once per week at the minimum.  We should probably put something on the wiki somewhere to answer this.

closed 2016-05-18 by rmackay9

alexblack, 2016-05-18:
Thanks for the response Randy, my question was about where to find the source, any suggestions?

rmackay9, 2016-05-18:
@alexblack, Ah ok.  I was perhaps jumping ahead.  In any case, here's where the POSITION_TARGET message comes in. https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1691

alexblack, 2016-05-18:
Cool, thanks @rmackay9 !