ArduPilot/ardupilot #4154: Question - Moving precisely without GPS

alexblack, 2016-05-18:
I'd like to be able to move and position the drone without GPS, I'm working on a visual positioning system that uses the camera to detect a target from which position can be determined.

I'm using `SET_POSITION_TARGET_LOCAL_NED` (through DroneKit) to move the drone (a 3DR Solo), and its generally working, but, the movements are not very precise, for example I attempted to move the drone 10cm, and it moved 48cm.

Questions:
1. Does `SET_POSITION_TARGET_LOCAL_NED` work without the use of GPS?
2. How should `SET_POSITION_TARGET_LOCAL_NED` be used for precise movements?

Currently I'm sending `SET_POSITION_TARGET_LOCAL_NED` every 100ms with a new velocity, on a curve, from 0 upto a max velocity and back to 0, where the integral of the curve is the distance I want to travel.

alexblack, 2016-05-18:
In particular altitude control seems poor, often it kinda of jumps up and down, could this be related to: http://diydrones.com/forum/topics/althold-inaccurate-altitude-estimation-and-motor-pulsing?xg_source=activity&id=705844%3ATopic%3A2170652&page=3#comments

magicrub, 2016-05-18:
You've posted a help question in the developer TODO list. Please re-post in
discuss.ardupilot.org

closed 2016-05-18 by magicrub

rmackay9, 2016-05-18:
It's also a duplicate of this one which I've answered.  https://github.com/ArduPilot/ardupilot/issues/4153

alexblack, 2016-05-18:
Hi Randy, oh, they're different questions, and I don't see an answer to the other one.. #4153 I asked where to find the source for SET_POSITION_TARGET_LOCAL_NED

alexblack, 2016-05-18:
Ok, thanks @magicrub