ArduPilot/ardupilot #4156: Frequency Interference preflight Check

Jman841, 2016-05-18:
#### Issue details
Yesterday I had some very dangerous situations when flying with a 1.5w 1.3ghz VTX on a plane with ardupilot

Details and logs are here, I have more logs from further flight tests if needed, also this log has 2 flights on it, the first is flown without the VTX and flies perfect, the second is the crazy behavior with the VTX:

 http://discuss.ardupilot.org/t/very-strange-unexplained-behavior-in-flight-please-help/9083

The issue is there was absolutely no warning from the ground that there was any issue as all preflight checks passed without issue, all control surfaces were working perfectly, and the GPS was showing solid lock. 

The other issue is, I do not know how to detect that there was an issue before take off. This interference resulted in very dangrous situations where the plane ends up stalling itself and it effects all automatic modes and FBWA. The strangest part is it makes the controller act crazy as the desired pitch and roll are always in line. If we can find some sort of erratic indicator of interference before take off, it could prevent a serious issue during flight.

#### Version
Arduplane 3.5.3

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [ X] Plane
- [ ] Rover

#### Airframe type
Twin engine traditional plane
#### Hardware type
3DR Pixhawk with 3DR GPS
#### Logs
Logs are in the link above.

magicrub, 2016-05-18:
I suspect gps jamming, I'll take a look at the logs in a couple hours when
I have a chance.

magicrub, 2016-05-18:
sat count was degraded during second flight, see forum post for details

closed 2016-05-18 by magicrub

Jman841, 2016-05-18:
I don't think this is resolved, the issue is definetly interference of the GPS and possibly other sensors, but the issue is that there was no preflight check to say there was a problem. Sat count was still at 9 or above which is a safe level to fly at, however, this resulted in a catastrophic flight if I didn't take over in manual mode. The issue is to somehow add a preflight check for GPS or other sensor interference to warn the pilot, of possible.

magicrub, 2016-05-18:
right now, in plane, all hell breaks loose when you lose GPS unexpectedly
in AUTO. Not sure what can be done about that pre-flight wise.

Jman841, 2016-05-18:
Not exactly sure as I do not understand the workings of it all. Any idea why FBWA would be effected as well?

The second graph wickedshell posted in the other discussion seems to show large interference almost right away when the VTX was plugged in. Maybe whatever that graph is showing should have a arming threshold? 

I am going to test tomorrow with a lower powered VTX in a position as far away as I can get it and hopefully it goes well, but, I wish there was a way to test before I take off.

WickedShell, 2016-05-18:
@Jman841 I'm double checking now, but I actually don't see us checking those anywhere in the arming checks, and I think we should be strongly checking them. The EKF has some thresholds before it accepts the data, so it makes sense to mirror those as a general arming check. I'm looking now and will open an issue on that aspect.

magicrub, 2016-05-18:
FBWA, much like other flight assisted modes, requires the EKF to not be in
a freak-out mode which is what happens when the GPS is giving bad data. I
think MANUAL is the only mode that is not using the EKF to direct the
aircraft.

Jman841, 2016-05-18:
Ah ok,

Thank you for the explanation. I have some other logs with the VTX installed from when I was trying to narrow down the problem. All the flights resulted in a stall.

Also, Tom, for the stall recovery side, these logs all have a stall and a manual recovery if it could be of any help for that as well. This plane recovers very easily considering its on the max flyable weight (1.8 meter foamy flying at 6 kg), but all of the recovers were done by switching to manual, adding power, and applying up elevator.

magicrub, 2016-05-18:
yes, please post a link to the logs in
https://github.com/ArduPilot/ardupilot/issues/2933 and
https://github.com/ArduPilot/ardupilot/issues/2669 with a short decription
of what happened and the aircraft type. More logs the better there, thank
you for sharing! I'm not sure when I'll get to those, I'd be perfectly
happy if someone else wants to work on it also. My TODO list is quite long
and the stall detection/recovery is somewhere in the middle

Jman841, 2016-05-19:
No problem, I'll post them when I get back to civilization. The internet out here is horrible.

iskess, 2016-05-21:
Would TRAINING mode be affected? It's a hybrid of MANUAL and FBWA. I ask because we use it as a manual take over mode since it provides full manual control with protections for over controlling, which often happens when the operator is urgently transitioning to manual flights with uncalibrated thumbs.

WickedShell, 2016-05-21:
It would be affected if your AHRS is being compromised as it might think
you are at limits when you aren't.

iskess, 2016-05-21:
That's what is thought. This is a serious vulnerability. We need to find a way to downgrade from EKF when it freaks out. Perhaps EKF is overkill for STABILIZE mode.  Is DCM an option?

Jman841, 2016-05-21:
I agree, I opened another issue for this, however, maybe the issue should be more broad to include all EKF errors, not just GPS errors.

New issue is here: https://github.com/ArduPilot/ardupilot/issues/4186

rmackay9, 2016-05-22:
I think the autopilot, no matter what, is going to be in bad shape if the attitude estimate (from the EKF) goes bad.  I don't think the solution is "let's switch to DCM" but rather to make sure the EKF is not vulnerable.  So really I think Paul Riseborough should have a look at the logs, if he says, "ah, no way the EKF could ever handle this" then the DCM option can be considered but I don't think he'll say that.
Adding additional pre-arm checks to ensure the EKF's attitude and/or position estimates are good before take-off is another good option.  Copter has 38 or so pre-arm checks and we're looking to merge those with AP_Arming lib (used by plane).  Maybe progressing more on that front would be a good solution.

iskess, 2016-05-22:
My thought was that using DCM for some basic modes like STABILIZE and CIRCLE would provide redundancy to EKF used in more complex modes. There is always some risk that EKF could fail, its smart to have a mode to fall back on with basic wing leveling for example.