ArduPilot/ardupilot #4179: TECS: Detect and handle overspeed conditions

WickedShell, 2016-05-20:
#### Issue details
If the aircraft is in an over speed condition TECS won't counter it past reducing the throttle, TECS will not demand a raise in pitch, which can lead to making the overspeed worse. Tagged as a safety feature since overspeeds can cause undo stress to the aircraft, or cause it to shed parts/lose control.

#### Version
3.5.x but still applies to master

#### Platform
- [ ] All
- [ ] AntennaTracker
- [ ] Copter
- [x] Plane
- [ ] Rover

#### Airframe type
Fixed wing

#### Logs
It can be seen in this log in the guided at the end. http://files.swiftradioplanes.com/30.BIN (CAUTION: 172MB DF log)

magicrub, 2016-05-20:
I haven't looked at the log yet but was the cause of the overspeed in the
waypoint mission?

WickedShell, 2016-05-20:
Was a large descent in a guided point where it was already at the point. ~200 meter change demanded, it had it under control, then didn't get quite the descent it wanted for a moment (thermals I suspect) which caused it to force the nose down to the limit (which by itself was fine), but as soon as it hit the max airspeed I think it should have backed off rather then continue to hold the max down pitch.

magicrub, 2016-05-21:
agreed. The TECS should always aim to keep the target airspeed within
the ARSPD_FBW_MIN/MAX flight envelope.

Naterater, 2019-01-04:
I just reviewed the parameters the other day, and it appears that the logic (behind the descriptions) would be to set the pitch to minimum without regard to ARSPD_FBW_MAX. Also on climbs, the pitch would not be reduced if the airplane was under TRIM_ARSPD_CM (even though tuning should make sure that that's not an issue). Unless anything has changed, I guess this is still a problem. I wonder if this is issue is partially related to the behavior on airspeed sensor failures (for the nose to pitch down, even when below altitude).