ArduPilot/ardupilot #4200: Plane: Programmable engine kill switch

naktinis, 2016-05-24:
In some situations it would be better to kill engine ignition instead of just suppressing the throttle, as this would work faster to take the propellers to a halt.

There is an option to set RCN_FUNCTION to "EngineRunEnable", but I don't see it ever being used in plane's code (only in AP_MotorsHeli_Single.cpp). There's even a comment next to "EngineRunEnable" saying "engine kill switch, used for gas airplanes and helicopters", although from what I understand it's not used anywhere in plane code.

magicrub, 2016-05-24:
instead of just suppressing the throttle, as this would work faster

Can you give us an example of this? The throttle suppression is automatic, are you proposing to have an additional one via RC input?

R-Lefebvre, 2016-05-24:
The idea when I started this, was that it was a secondary kill system (typically ignition system) that would be tied to armed/disarmed status.  This is because some regulations require two kill methods for large gas motors.  And this way, the user would *not* be required to use a separate channel to control the kill, it would be automatic.

I don't think the work got finished though.

gelazanskas, 2016-05-25:
What Rimvydas and I want to implement is automatic ignition kill 1) after landing, 2) before [emergency] parachute release, 3) when operator wants to do so (manual kill). For carburettor engines it is currently done by lowering the throttle servo until engines "suffocates" and dies. In addition, one can implement ignition kill switch, which cuts off voltage to the spark and kills the engine faster (we do not need to wait for the remaining droplets of fuel to burn out in the engine). This kill switch is controller by PWM. We believe there should be functionality (through RCx_FUNCTION) that would control the kill switch.

R-Lefebvre, 2016-05-27:
Yep, exactly.

And some regulations require an ignition kill method.  I believe Tridge's flying club requires this, and I think he's doing it now via a channel passthrough.  IMO, it would be safer to tie this to an automatic method.

jinchengde, 2016-05-31:
May I ask then we could cut off the engine with shut down the CDI for gas engine?

magicrub, 2016-05-31:
Via disarm

DavidIngraham, 2017-11-30:
FYI, this is currently supported through the AP_ICE Library. It is currently undocumented, but it does work. However, it only supports a PWM output, not a relay output, so it requires an external RC switch module to cut power to the ignition module.

peterbarker, 2024-05-07:
We now have relay support for ignition in AP_ICEngine.

I think this issue's resolved now.  Please re-open if that's wrong.

closed 2024-05-07 by peterbarker