ArduPilot/ardupilot #4212: Copter pulses on Motor test

lvale, 2016-05-26:
#### Issue details
While performing motor test via MP all the motors twitch a little when a motor test button is clicked
#### Version
3.4master 25-5

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad
#### Hardware type
PixRacer

#### Logs
N/A

Already discussed and detected that a pulse is sent to all the motors

tridge, 2016-05-26:
Hi Luis, I think this is fixed now. Can you please test master again on your quad?
I'd also like to know if OneShot125 is working for you again

lvale, 2016-05-26:
Stranger...

Cleared everything, redid calibrations (radio and accels) and I've setup the parameters as in the file attached.

Attached logic analyses reflects the following actions:
Plugin battery, Wait for the ready tone on the Racer, press the safety button, motors start twitching, even not armed. Armed and raised throttle to max twice and motors still twitching. Disarmed and motors still twitching. Pressed safety switch and motors keep twitching. 

Only way to stop is to disconnect battery.

[GH issue.zip](https://github.com/ArduPilot/ardupilot/files/283394/GH.issue.zip)

lvale, 2016-05-26:
https://github.com/ArduPilot/ardupilot/commit/53fc095d4ce6ac0783acfb7394d3187502a43530

should be enough to address issue

closed 2016-05-26 by lvale