ArduPilot/ardupilot #4214: Fix implementation of _parse_decimal_100() in AP_GPS_NMEA

jaluna, 2016-05-26:
#### Issue details
_Please describe the problem, or desired feature_

AP_GPS Library, AP_GPS_NMEA.cpp needs to fix parse_100 (as in Master branch)

See 
https://github.com/ArduPilot/ardupilot/pull/3602#event-666440090
https://github.com/ArduPilot/ardupilot/pull/4166#event-667152498

#### Version
_What version was the issue encountered with_
3.3.3
#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
hex
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2016-05-26:
Beta testing of 3.4 will start soon, and since this isn't exactly a safety issue, I don't think we will release a new 3.3 version to fix this. @rmackay9?

jaluna, 2016-05-27:
I don't want to enter in the discussion of if its a safety issue or not, but we have a fleet of 20 drones flying in formation (an show), so we have fixed in below way and we have solved the problem of automatic landing.

in h
int32_t    _parse_decimal_100();

in cpp
#define round(x)								((x)>=0?(long)((x)+0.5):(long)((x)-0.5))

int32_t AP_GPS_NORTH::_parse_decimal_100()
{	
    char *q, *p = _term;
		float frac = 0;
		int sign = ((*p == '-')? -1: 1);
    int32_t ret = 100L * atol(p);
    while (isdigit(*p) || *p == '-')
        ++p;
    
    // convert fractional part
    if (*p == '.') {
        q = p + 1;
        float frac_scale = 0.1f;
        while (*q && isdigit(*q)) {
            frac += DIGIT_TO_VAL(*q) * frac_scale;
            q++;
            frac_scale *= 0.1f;
        }
    }
    return ret+sign*round(frac*100L);
}

I hope it helps.

closed 2016-05-27 by jaluna