ArduPilot/ardupilot #4218: No inertial speed in MAVlink flow

night-ghost, 2016-05-27:
#### Issue details

There is no  inertial speed in MAVlink flow, so in GPS-less setups a lots of speeds always ==0. There is inertial_nav.get_velocity_xy() function that used internally in 2 places but not for MAVlink. 

I understand that this speed is very inaccurate, but it is better this than nothing. So when gps_fix_type==0 (that means no GPS) airspeed and groundspeed in MAVlink should be set to inertial_nav.get_velocity_xy() normalized to the right units.

PS. Yes, for me too, the existence and popularity of such setups was a surprise, but a few people with 120-250 size copters complained that in the absence of GPS OSD does not display neither height nor speed.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

peterbarker, 2021-07-30:
@Hwurzburg do you know if we get this data in OSDs now?

Hwurzburg, 2021-07-30:
no, sorry....dont fly multi without GPS...guess you can now without GPS OR BARO....
guess we would have to modify the calls instead of using ahrs.groundspeed_vector() and ahrs.get_relative_position_D_home(alt)
inertial nav has got to be really flaky without GPS/BARO, especially on a small quad going all over the place...no?