ArduPilot/ardupilot #4225: Fused Altitude reading always 0 on Solo w/functional Lidar Lite

a7scenario, 2016-05-27:
#### Issue details
Running Master from 26 May on a 3DR Solo w/Lidar Lite (blue label) and PX4Flow.  When airborne, the the Altitude in Mission Planner and in DroneKit show 0 in the GCS HUD and in VFR_HUD messages.  The RANGEFINDER messages show correct altitude (3-4 meters) and I've confirmed that both EK2_ALT_SOURCE=1 (use rangerfinder not baro) and EKF_ALT_SOURCE=1, however, the Solo is running EK2.  

#### Version
Master from 26 May

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
3DR Solo

#### Hardware type
Pixhawk 2

#### Logs

rmackay9, 2016-05-30:
I think I've heard this report somewhere else as well although it was in a different context.  Perhaps the vehicle didn't have a GPS lock yet.. or to be more precise the EKF hadn't selected it's "origin" yet so there was not home location set (i.e. the vehicle wouldn't have shown up on the ground station map yet).  In this case, it seems we're not updating the altitude sent to the GCS.

SimsGautam, 2016-07-03:
Hello -- bumping this issue, as I'm having a similar problem where the fused altitude reading is either 0 or incorrect (AHRS2 always reads zero, and AHRS3 is faulty). My teammate has written more extensively on the issue here: http://discuss.ardupilot.org/t/guided-mode-with-optical-flow-oscillating-in-altitude/9992

Thanks!