ArduPilot/ardupilot #4226: Compile failed with gcc 6.1 on Ubuntu 16.04

khancyr, 2016-05-27:
#### Issue details
@tridge 
Hi, I use gcc version 6.1.1 20160510 (Ubuntu 6.1.1-2ubuntu12~16.04) on Ubuntu 16.04 and I got new error because of this commit https://github.com/ArduPilot/ardupilot/pull/4202: 
````
../../libraries/AP_GPS/AP_GPS_ERB.cpp: In member function ‘virtual bool AP_GPS_ERB::read()’:
../../libraries/AP_GPS/AP_GPS_ERB.cpp:113:47: error: array subscript is above array bounds [-Werror=array-bounds]
                 _buffer.bytes[_payload_counter] = data;
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

../../libraries/AP_Mount/AP_Mount_SToRM32_serial.cpp: In member function ‘void AP_Mount_SToRM32_serial::read_incoming()’:
../../libraries/AP_Mount/AP_Mount_SToRM32_serial.cpp:239:39: error: array subscript is above array bounds [-Werror=array-bounds]
         _buffer.bytes[_reply_counter++] = data;
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

../../libraries/AP_Mount/AP_Mount_Alexmos.cpp: In member function ‘void AP_Mount_Alexmos::read_incoming()’:
../../libraries/AP_Mount/AP_Mount_Alexmos.cpp:270:51: error: array subscript is above array bounds [-Werror=array-bounds]
                     _buffer.bytes[_payload_counter] = data;
                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors

../../libraries/AP_GPS/AP_GPS_UBLOX.cpp: In member function ‘virtual bool AP_GPS_UBLOX::read()’:
../../libraries/AP_GPS/AP_GPS_UBLOX.cpp:434:47: error: array subscript is above array bounds [-Werror=array-bounds]
                 _buffer.bytes[_payload_counter] = data;
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
compilation terminated due to -Wfatal-errors.
cc1plus: some warnings being treated as errors
````

#### Version
Master

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

khancyr, 2016-06-01:
@tridge 
array of length zero doesn't work ... could we revert to array of length 1 ?

tridge, 2016-06-05:
it is building fine with gcc 6.1.0 here (I built the 6.1.0 release from source to test it)

khancyr, 2016-06-07:
I used gcc 6.1.1 for ubuntu ppa https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test
Your patch doesn't work, reverting it works ... don't know why... ...
Got to many errors so i revert to gcc5.4, those problems are too hard to solve for me sorry.

By the way, to see zero lenght array and flexible array error, we need to use -pedantic flag , which print many other warning too ahhahahah

lucasdemarchi, 2016-11-08:
Building fine here, must have been fixed. Let me know if it wasn't.

closed 2016-11-08 by lucasdemarchi

khancyr, 2016-11-08:
Yep it was solved by the pr referenced here !