ArduPilot/ardupilot #4230: Can ArduPilot use WIND_COV from common message set intead of ardupilot dialect WIND message?

DonLakeFlyer, 2016-05-28:
This would make it easier to write a single bit of GCS code.

WickedShell, 2016-05-31:
Consensus is that this seems totally reasonable and we can mark the ardupilot wind as deprecated and send both for awhile.

However in the WIND_COV are the wind_x etc expected as to or from's? (And the XML should be made more specific)

And for things that we don't have a value for (accuracies for example) do we simply report 0 or is there some other accepted value? (and again we should update the XML to indicate this)

peterbarker, 2026-11-15:
I've just had a look, and ArduPilot sending this according to spec would not actually be helpful to anybody - at the moment it would really just muddy the waters.

PX4 is the only publisher I've found, and it fills the fields in incorrectly - it puts variances (roughly speaking, it's worse than that) in instead of 1-std wind measurement deviations.  MAVROS *consumes* those fields as variances - so they're basically talking the same language, it's just not that specified in the mavlink message!

MAVSDK tries to interpret these according to the spec... perhaps it should omit them from any presentation in the SDK (@julianoes )

`WIND` isn't entirely fit for purpose either as the vehicle may have multiple estimates of it.  But IIWII for now, and I do note that QGC has support for it.  And bandwidth isn't scaling as much as we'd like :-)

julianoes, 2026-11-16:
Happy to change/adapt MAVSDK either direction once we agree.