ArduPilot/ardupilot #4235: GPS: SBF is continously detected and then reported as lost when not connected

WickedShell, 2016-05-31:
#### Issue details
Every 2 seconds it reports detecting the GPS as SBF (when there is no GPS connected)

#### Version
e7974702 (but still valid for master)

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Searching for SBF GPS

meee1, 2016-05-31:
this is the gps autodetect code at work. it is cycling through baud rates, and if it doesn't get a valid response, the autodetect code moves on to the next baud rate

the GSOF/SBP and soon to be NOVA drivers all do this.

WickedShell, 2016-05-31:
I get that its autodetect, but this means to the ground/external side that we have found a valid SBF instance (which is why auto detect passed). If there was never a response how is it being accepted?

meee1, 2016-05-31:
there is no autodetect sequence for these gps's. they are always manually specified

WickedShell, 2016-05-31:
Does that mean that the SBF driver manages baud rate? As everytime we restart autodetect we go back to the first baud rate there?

meee1, 2016-05-31:
no, the autodetect managed the baudrate.
1. autodetect sets baud
2. autodetect inits driver
3. driver inits gps (or ties to)
4. if baud was bad, gps doesn't init, move to next baud.
5. baud is correct, gps inits.
6. valid message received, continue as normal.

should the gps have issues in flight, this means that the autodetect code would reinit the gps, if no valid messages are received.

meee1, 2016-05-31:
its using the autodetect code to autobaud effectively.

WickedShell, 2016-05-31:
I'm sorry, I thought when we unloaded a driver we reset it's detect state, I see thats not actually the case. Maybe we should just move the print of detecting the driver inside of the SBF layer rather then in the autodetect level?

IE: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp#L241 inside AP_GPS_SBF that way its not always happening while searching? (and only once we've positively ID'd a GPS as SBF)

meee1, 2016-05-31:
yep, makes sence. and prevents the spamming

amilcarlucas, 2017-09-25:
@WickedShell has this been addressed in your latest SBF PR ?

amilcarlucas, 2020-11-23:
I stil have this with a Septentrio Asterx m2-UAV running FW 4.8.0 connected to a CubeBlack running ArduCopter 4.0.5 :(
I can only save the SBF configuration as boot configuration, if I get ardupilot to see the GPS at least once.

peterbarker, 2021-07-30:
I'm going to go with "yes, this has been fixed", or we'd have heard more complaints by now....

WickedShell, 2021-07-30:
Nope, absolutely wrong.