ArduPilot/ardupilot #4240: We've had 3 different vehicles have flyaway behavior since moving to 3.3

piningforfjords, 2016-06-01:
#### Issue details
  This is my first bug report.  We have been developing a machine for 1.5 years and have had amazing experience with APM/Pixhawk.  The move to 3.3 was equally amazing until we hit this bug.  We have had 3 different vehicles become completely uncontrollable since moving to 3.3.   All these vehicle have had extensive testing for months.  Afterwards they were flown via RC and showed no problems and flew with Mission planner with no trouble.  The code that initiated the problem is a mav proxy module that we have written that submits a flight plan and launches the vehicle.  We have the option to have RC connected but in the first case (Black machine) it was not.  In the second case rc was connected but was completely unresponsive.  We did find a bug in our code that entered non sequential line numbers in the flight plan but this does not seem to cause a problem for the autopilot in subsequent testing.  I've spent a month trying to reproduce the problem without success.
  In the first case (Black) the vehicle just kept climbing and drifted down wind until max altitude was reached and it stated to come down.  
  In the second case (Blue) the vehicle started to climb and using RC the vehicle was taken out of auto mode and it just dropped like a stone.  No amount of throttle made a difference.  That experiment was repeated with the same results.  We have had a 3rd machine do the exact same thing. I don't have the logs for that one.   We make very durable machines and they have flown subsequently with no repair work and show no adverse problems or vibration.
  I have unsuccessfully tried to reproduce the problem with tethered flights launched from our MavProxy module.  I've also ran batch uploading and downloading of flight plans to see if the flight plan comes back with any changes (due to renumbering the flight plan).
  In the logs you can see regions of heavy vibration but I believe (conjecture) this is actually caused by the flight controller.  The same vehicles still fly beautifully in strong wind and calm weather.

#### Version
3.3

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad H Frame

#### Hardware type
Dropix (pixhawk clone)

#### Logs
Black #1st flight
https://www.dropbox.com/s/a3vgbborfxhgk30/Black%20-%20Fly%20Away.bin?dl=0
https://www.dropbox.com/s/mfyrsa4q9k9hvzs/loadAusflightThenReadVehicleFP.png?dl=0

Blue #1st flight
 https://www.dropbox.com/s/0z5jbe0rxaq9rif/Blue%20-%20Fly%20Away%20with%20RC%20abort.bin?dl=0

Blue #2nd flight 
https://www.dropbox.com/s/t5r7taz0lt3zlzm/Blue%20-%20Fly%20Away%20with%20RC%20abort%202.bin?dl=0

An example of a poorly numbered flight plan generated by our mavproxy module
https://www.dropbox.com/s/go5exy9powobuna/way.txt?dl=0

piningforfjords, 2016-07-30:
Ok I have some new info on this.  I have been able to reproduce this problem once in probably a hundred takeoffs.  I haven't had a chance to test this further but it looks like this problem crops up when the vehicle is is started up in a new location.  On the first boot in this new location it seems like there is a bigger chance for this flyaway behavior to occur.  I don't believe it happens on subsequent boots in the new location.  It would be very helpful in reproducing this if there was a way to tell the vehicle it was in another location before shutting it down.

magicrub, 2016-07-31:
I haven't looked at your logs but check that your arming checks are looking at ekf and gps. I've only heard of fly aways when you take off before it was ready, like it's trying to loiter at position 0,0 or something.

piningforfjords, 2016-08-01:
Hi Tom

  Thanks for replying.  That is very helpful indeed.  I had been doing GPS
checks separate to ensure lock and a stable position but I didn't realize
EKF was loaded onto the same check in 3.3.  The vehicles all headed East
South East which would show a desire to 0,0 from where I am so I would say
that is probably the issue.  Many thanks for the insight.

James

closed 2016-08-01 by magicrub

magicrub, 2016-08-01:
they call them arming checks for a reason :) the default of check_all is
highly recommended.

piningforfjords, 2016-08-15:
> issuecomment-236397712
> AGYnr7qWMg6V7cZwpBV1TFP7UZCMXxI4ks5qa-aCgaJpZM4IrKI4

Hi Tom

  One question.  Is there something in the logs that I could use to verify
that this was indeed the cause?  I recognize the importance of the checks
and in general they are used but we do some indoor flight testing that
requires the GPS be turned off.  Also we have a more strict ground station
GPS check so we did get sloppy on that one.  I wasn't aware that EKF check
was overloaded on that setting.

Thanks again
Tom

magicrub, 2016-08-15:
Im no expert at Copter logs, I'm a plane guy, but you can browse the values ahrs and gps sections to see what it thought it's current location was.