ArduPilot/ardupilot #4242: Copter: 3.4RC1 safety switch

lvale, 2016-06-01:
#### Issue details
Safety switch action

#### Version
3.4RC1
#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Copter Quad
#### Hardware type
PixRacer
#### Logs

WickedShell, 2016-06-01:
Any more details? I have no clue what this is about...

lvale, 2016-06-01:
Safety switch is ignored. I'm in the process of retrieving logs.

magicrub, 2016-06-01:
Doh!

magicrub, 2016-06-01:
I made a change to this recently so I must have broke it. Sorry!

magicrub, 2016-06-01:
I just tested it in plane and it works fine. Can someone else test master
on copter and report back?

lvale, 2016-06-01:
What was the change that caused the issue ?? I can try it as soon as the build server has a copter-quad version, my compilation environment is broken :)

magicrub, 2016-06-01:
This commit:
https://github.com/ArduPilot/ardupilot/commit/67aaf7e226fd1ac9d6acd3589a53e962dab91588

lvale, 2016-06-03:
I was talking with Tom, and the issue is NOT common to all the FC. It only happens on the PixRacer.

rmackay9, 2016-10-15:
This is fixed now I think right?

closed 2016-10-15 by rmackay9