ArduPilot/ardupilot #4246: AntennaTracker not forwarding Mavlink to GCS in master

TimothyGold, 2016-06-01:
After changes in the Mavlink libraries over the last several weeks and changes in GCS_Mavlink.cpp I noticed that AntennaTracker in master is no longer forwarding packets to the GCS.  In Mission Planner the Antenna Tracker is the only ID that is discovered.

Version - 
ArduPilot/ardupilot/master/AntennaTracker

#### Platform
[X] AntennaTracker

rmackay9, 2016-06-02:
This is fixed in master now.  Tridge fixed the issue a few days ago.  I think you'll find the newest beta works again.  I'll close this but please tell me if you think I'm wrong.

closed 2016-06-02 by rmackay9

TimothyGold, 2016-06-02:
Last night I was able to merge the Ardupilot/master into my fork.  I was able to compile and try it, but I ran into the same result. In the GCS only Tracker shows up.  I will try again from the main repository of Ardupilot/master tonight.

rmackay9, 2016-06-02:
@TimothyGold, ok.  When we first pulled in Tridge's fix we also didn't see the vehicle.  We had to do a "make px4-clean" and perhaps a "submodule update --recursive" to make sure we got the fix which is actually in mavlink.
Just to be 100% sure it's not a build problem, I think loading up the 0.7.7 beta using Mission Planner would be best (the binary can also be found on firmware.ardupilot.org).

TimothyGold, 2016-06-02:
@rmackay9, that was it.  I did do a clean build but I completely forgot about the potential submodules that I would need to update.  Thanks for your help.