ArduPilot/ardupilot #4247: Motors spin very fast after restart of ArduCopter @Navio2

dgrat, 2016-06-01:
#### Issue details
It happened just one time so far, after several trials. 
Restarting ArduCopter may lead to odd behavior like sudden spin-starts of the motors. 

#### Version
3.4 beta

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
Navio2

#### Logs
Try to make one

staroselskii, 2016-06-01:
@dgrat 

It's possible to log before arming. You just need to add ```DisarmedLogging``` flag in GCS!

staroselskii, 2016-06-03:
@dgrat 
Could you please add information about rcio-dkms package that you're using? That's a crucial piece of information.
I'm also wondering what kind of restart you're talking about? Is it the immediate one that was triggered by systemd or a manual one? (or it doesn't matter).

I was trying to reproduce your issue but to no avail :-( I plugged my Saleae Logic analyzer and restarted ArduCopter a dozen times. I haven't noticed anything strange at all. Is there a chance that your ESCs are misconfigured and faulty?

dgrat, 2016-06-04:
@staroselskii 
Thanks, 
  - I tested with: "sudo killall arducopter-quad && sudo ./arducopter-quad" but it also happens if you restart via systemd.
  - I use the rcio driver that came via the RPi3 image. I will also test it with the current one later.
  - My ESCs seem fine, I use HobbyWing XRotor 50 A
I think I will try to get more info about this, and post it here. 
Maybe I can try to write a demo which can reproduce this in 100% of cases next week.

staroselskii, 2017-04-19:
This is not ArduPilot issue per se but rather a rcio-dkms driver. So I'm closing this one.

closed 2017-04-19 by staroselskii