ArduPilot/ardupilot #4249: SITL Floating Point Error with Plane 3.6 beta1

Jesse-Millwood, 2016-06-01:
#### Issue details
I got a Floating point error while running ArduPlane in SITL. I ran mavproxy with the -G option so here is the back trace: 
```
Program received signal SIGFPE, Arithmetic exception.
AP_Mission::mavlink_to_mission_cmd (packet=..., cmd=...)
    at /home/acusar/build/ardupilot/libraries/AP_Mission/AP_Mission.cpp:863
863             mav_cmd.y = packet.y * 1.0e7f;
(gdb) bt
#0  AP_Mission::mavlink_to_mission_cmd (packet=..., cmd=...)
    at /home/acusar/build/ardupilot/libraries/AP_Mission/AP_Mission.cpp:863
#1  0x000000000045f068 in GCS_MAVLINK::handle_mission_item (
    this=0x730450 <plane+7664>, msg=0x7fffffffdbf0, mission=...)
    at /home/acusar/build/ardupilot/libraries/GCS_MAVLink/GCS_Common.cpp:742
#2  0x000000000040db42 in GCS_MAVLINK::handleMessage (
    this=this@entry=0x730450 <plane+7664>, msg=msg@entry=0x7fffffffdbf0)
    at GCS_Mavlink.cpp:1787
#3  0x000000000045d948 in GCS_MAVLINK::update (
    this=this@entry=0x730450 <plane+7664>, run_cli=...)
    at /home/acusar/build/ardupilot/libraries/GCS_MAVLink/GCS_Common.cpp:975
#4  0x000000000040d395 in Plane::gcs_update (this=0x72e660 <plane>)
    at GCS_Mavlink.cpp:2196
#5  0x000000000047da1f in Functor<void>::operator() (this=<optimized out>)
    at /home/acusar/build/ardupilot/libraries/AP_HAL/utility/functor.h:55
#6  AP_Scheduler::run (this=0x72e9b8 <plane+856>, time_available=20000)
    at /home/acusar/build/ardupilot/libraries/AP_Scheduler/AP_Scheduler.cpp:141
#7  0x00000000004e3c2e in HAL_SITL::run (
    this=0x7325e0 <AP_HAL::get_HAL()::hal>, argc=<optimized out>, 
    argv=<optimized out>, callbacks=0x72e660 <plane>)
    at /home/acusar/build/ardupilot/libraries/AP_HAL_SITL/HAL_SITL_Class.cpp:95
#8  0x0000000000402fce in main (argc=<optimized out>, argv=<optimized out>)
    at ArduPlane.cpp:1015
```
This occurred while I was trying to upload a mission. 
I did find the issue however, which happened to be my error by accidentally forgetting the decimal point making the lat -104845746 instead of -104.845746. Once I fixed that, all worked fine.
I just thought I would throw this out there though. I suppose it might not be that big of a deal since its way out of the proper range and a proper gcs would probably not send garbage like that.

#### Version 3.6 beta1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane (Only one tested)
[  ] Rover

#### Airframe type: fixed wing

#### Hardware type: SITL

magicrub, 2016-06-01:
Hi Jesse, thanks for the bug report. There is a sanity check for valid GPS
bounds in there but this must not be using it. We'll get that fixed up for
the next beta

Jesse-Millwood, 2016-06-01:
Great! 
If it helps, the frame I was using for the mission item was GLOBAL.

WickedShell, 2016-06-01:
ARGH! There is but its all encoded in the mission_int impl, and the FPE is in the process of getting to a mission_int... Easy to fix up though....

WickedShell, 2016-06-06:
#4279 fixed

closed 2016-06-06 by WickedShell