ArduPilot/ardupilot #4250: Quadplane: is_flying=true is inhibiting RC disarm

magicrub, 2016-06-01:
#### Issue details
Quadplane: is_flying=true is inhibiting RC disarm. 
Offending code lines are:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/radio.cpp#L136
and
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/is_flying.cpp#L31

#### Version
Plane v3.6.0beta1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover

#### Airframe type
quadplane

#### Hardware type
pixhawk

magicrub, 2016-06-01:
The problem is not in radio.cpp. It is the is_flying=true override for quadplane

tridge, 2016-08-29:
this was fixed with the update to is_flying in quadplane 
you can now rudder disarm in quadplane modes if at zero throttle

closed 2016-08-29 by tridge