ArduPilot/ardupilot #4254: waf px4-v2 build fails on fresh clone of ardupilot

peterbarker, 2016-06-02:
#### Issue details
Problems occur in PX4Firmware when doing the following sequence:

```

cd tmp
git clone https://github.com/ardupilot/ardupilot APM
cd APM
git submodule init
git submodule update
vagrant up
vagrant ssh
cd /vagrant
./modules/waf/waf-light configure --board px4-v2
./modules/waf/waf-light examples

```

#### Version
master

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
n/a

#### Hardware type
n/a

#### Logs

```

vagrant@vagrant-ubuntu-trusty-32:/vagrant$ ./modules/waf/waf-light configure --board px4-v2
Setting top to                           : /vagrant 
Setting out to                           : /vagrant/build 
Setting board to                         : px4-v2 
Using toolchain                          : arm-none-eabi 
Checking for 'g++' (C++ compiler)        : arm-none-eabi-g++ 
Checking for 'gcc' (C compiler)          : arm-none-eabi-gcc 
Checking for HAVE_CMATH_ISFINITE         : no 
Checking for HAVE_CMATH_ISINF            : no 
Checking for HAVE_CMATH_ISNAN            : no 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for program 'cmake'                   : /usr/bin/cmake 
Checking for program 'ninja, ninja-build'      : not found 
Checking for program 'cp'                      : /bin/cp 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version                    : (2, 7, 6, 'final', 0) 
Checking for program 'git'                     : /usr/bin/git 
Gtest                                          : PX4 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
'configure' finished successfully (3.739s)
vagrant@vagrant-ubuntu-trusty-32:/vagrant$ ./modules/waf/waf-light examples
Waf: Entering directory `/vagrant/build/px4-v2'
[1/4] Running Submodule update: PX4Firmware
[3/4] Running Submodule update: uavcan
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule path 'modules/uavcan/dsdl': checked out '9804a3e6972825586be252ce08dd899f44994b14'
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
[... 116 lines cut ...]
Note: message HIL_STATE_QUATERNION is longer than 64 bytes long (72 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message LOG_DATA is longer than 64 bytes long (105 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_INJECT_DATA is longer than 64 bytes long (121 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message SERIAL_CONTROL is longer than 64 bytes long (87 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message ENCAPSULATED_DATA is longer than 64 bytes long (263 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message RESOURCE_REQUEST is longer than 64 bytes long (251 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message FOLLOW_TARGET is longer than 64 bytes long (101 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message CONTROL_SYSTEM_STATE is longer than 64 bytes long (108 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message AUTOPILOT_VERSION is longer than 64 bytes long (68 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message GPS_RTCM_DATA is longer than 64 bytes long (190 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Note: message V2_EXTENSION is longer than 64 bytes long (262 bytes), which can cause fragmentation since many radio modems use 64 bytes as maximum air transfer unit.
Merged enum MAV_CMD
Found 182 MAVLink message types in 2 XML files
Generating C implementation in directory /vagrant/build/px4-v2/libraries/GCS_MAVLink/include/mavlink/v2.0/ardupilotmega
Generating C implementation in directory /vagrant/build/px4-v2/libraries/GCS_MAVLink/include/mavlink/v2.0/common
Copying fixed headers for protocol 2.0 to /vagrant/build/px4-v2/libraries/GCS_MAVLink/include/mavlink/v2.0

Waf: Leaving directory `/vagrant/build/px4-v2'
Build failed
 -> task in 'px4_msg_gen' failed (exit status 2): 
	{task 3067601452L: cmake_build_task  -> }
['/usr/bin/cmake', '--build', '/vagrant/build/px4-v2/modules/PX4Firmware', '--target', 'msg_gen']
vagrant@vagrant-ubuntu-trusty-32:/vagrant$ 

```

OXINARF, 2016-06-02:
It has an old version of CMake, you need, at least, 3.2

peterbarker, 2016-06-02:
Thanks!
   
I guess we're going to have to move the Vagrant VM image forward to Ubuntu 16.04
- that make sense @tridge?
   
Second question is the autotest server (@davidbuzz)

autotest@ruki:~$ cmake --version
cmake version 2.8.12.2
autotest@ruki:~$

guludo, 2016-06-02:
Hm, I think it's a good idea to require a minimal version of cmake in the configure step.

lucasdemarchi, 2016-06-02:
you can require Ubuntu 16.04 or install cmake from ppa: https://launchpad.net/~george-edison55/+archive/ubuntu/precise-backports

guludo, 2016-06-07:
I'm closing this. :-)

closed 2016-06-07 by guludo