ArduPilot/ardupilot #4255: NMEA fix - too small to fork

adorus, 2016-06-02:
#### Issue details

in  ardupilot/libraries/AP_GPS/AP_GPS_NMEA.cpp 

following code has a bug:

№97 #define hexdigit(x) ((x)>9?'A'+(x):'0'+(x))

I bet this should be like this:

№97 #define hexdigit(x) ((x)>9?'A'+(x-10):'0'+(x))

#### Version
all versions

#### Platform
[ *] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
all

#### Hardware type
NMEA GPS receivers

#### Logs
none

magicrub, 2016-06-02:
Thanks @adorus , you are correct. However, that causes the following compile error:
> ardupilot/libraries/AP_GPS/AP_GPS_NMEA.cpp:468:7: error: right shift count is negative [-Werror]

which is: 
> if (hexdigit(state.ck>>4) == data) 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_NMEA.cpp#L468

Seems the state machine logic has a problem there

adorus, 2016-06-03:
Would it work that way?:
№97 #define hexdigit(x) ((x)>9?'A'-10+(x):'0'+(x))

magicrub, 2016-06-03:
That should fix the warning but I'm wondering how this happened in the
first place.  It clearly should not be able to parse a string right now so
clearly this is untested code.

Do you have an nmea gps to test this?

adorus, 2016-06-03:
I have a receiver. It can work in NMEA mode which is default for it. Unfortunately I have not any compatible pilot hardware yet. A Pixhawk box should arrive to my place in one month approx. Will try to test all this then.

magicrub, 2016-06-03:
If you don't mind me asking, how did you find this bug? Browsing source?

OXINARF, 2016-06-03:
@magicrub The _hexdigit_ was introduced by @tridge, maybe he can give some input here?

adorus, 2016-06-03:
Browsing source. I decided to grasp how pilot works while waiting for hardware.

tridge, 2016-06-05:
yep, the fix is correct. Well spotted!
It worked as it would end up just detecting the next msg OK, but better to fix it. Pushed.

closed 2016-06-05 by tridge