ArduPilot/ardupilot #4257: Use telemetry to control landing gear etc.

sashan13, 2016-06-02:
#### Issue details

I have 8 channel radio, and I find that I sometimes need a few more. However, I already have telemetry connected to my phone attached to radio. 

What about adding an option to toggle landing gear and perform other similar on/off actions using the telemetry on Android? 

Also, writing a message about gears up/down and other events to GCS would be good. MinimOSD can display that on screen, which would be incredibly useful when flying FPV. 

#### Platform
Copter
Plane

ferret-guy, 2016-06-02:
This needs to be implemented in tower, we already have a [DO_SET_SERVO](https://pixhawk.ethz.ch/mavlink/#MAV_CMD_DO_SET_SERVO) command that can be triggered from mission planner

rmackay9, 2016-06-03:
Sounds like a good idea to add this.  We would need to do a few things to do this:
1. decide whether to use do-set-servo or perhaps create a new MAV_CMD_DO_LANDING_GEAR (I think I prefer to add this new enum)
2. add support to GCS_MAVlink.cpp in each vehicle (or at least copter and plane).

ferret-guy, 2016-06-03:
I like the idea of MAV_CMD_DO_LANDING_GEAR becouse it does not require users to configure on every GCS they use what channel they are using, we could even change to a servo bitmask for landing gear for people using more than one channel, additionally this would be a useful addition to auto missions

lvale, 2016-07-04:
DO_SET_SERVO does the job perfectly.

squilter, 2016-07-05:
@lvale as ferret-guy said, `DO_SET_SERVO` requires the user to configure the channel and the pwm value on each GCS. This would seem more appropriate to configure in the vehicle parameters, and then just send a simple message. It would also be good if the vehicle could report to the GCS if the landing gear was retracted or not.

sashan13, 2017-02-10:
I guess this was abandoned. Any chances it will be added?

peterbarker, 2017-12-19:
See MAV_CMD_AIRFRAME_CONFIGURATION

DavidIngraham, 2018-04-26:
This is addressed in https://github.com/ArduPilot/ardupilot/pull/7893

Naterater, 2019-01-05:
Considering that landing gear is now supported, I'm assuming it is up to the GCS to provide an interface to manipulate the landing gear. Is that correct? If so, this should be closed and an issue for a GCS should be created.

peterbarker, 2021-07-30:
Landing gear can be manipulated using the run-aux-function command from a GCS.

Closing this

closed 2021-07-30 by peterbarker