ArduPilot/ardupilot #4260: SERVO_TYPE

Georacer, 2016-06-03:
#### Issue details
No description text is given for the SERVO_TYPE parameter. Should be something similar to 0:Angle, 1:ON/OFF, 2:Continuous Rotation.

It might be related, but in the Extended Tuning screen of Mission Planner, the dropdown menu for the corresponding relation is also blank.
[image]
[image]

#### Version
v0.7.6

#### Platform
[  ] All
[ X] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
3DR Pixhawk

#### Logs
N/A

magicrub, 2016-06-03:
The parameter descriptions only update on Mission planner when that
platform has an official release.

<https://cloud.githubusercontent.com/assets/3149996/15769783/fccae5b2-2966-11e6-9466-862e0e9c86ad.png>
[image: servo_type_param_2]
<https://cloud.githubusercontent.com/assets/3149996/15769788/045dd802-2967-11e6-9783-ff72787988fc.png>

magicrub, 2016-06-03:
I think there may be a tracker release coming out soon.  If it doesn't
update then please post an issue on the Mission planner issue list.
On Jun 3, 2016 12:08 AM, wrote:

The parameter descriptions only update on Mission planner when that
platform has an official release.

<https://cloud.githubusercontent.com/assets/3149996/15769783/fccae5b2-2966-11e6-9466-862e0e9c86ad.png>
[image: servo_type_param_2]
<https://cloud.githubusercontent.com/assets/3149996/15769788/045dd802-2967-11e6-9783-ff72787988fc.png>

closed 2016-06-03 by magicrub

Georacer, 2016-06-03:
I verify that it works with 0.7.7 beta. The new parameters for yaw and pitch servo are visible and selectable.

As a side bonus, I can say that the control of continuous rotation servos works fine.

rmackay9, 2016-06-03:
Thanks very much for testing the tracker.  I'm pretty sure this is a mission planner issue.  We might need to check with MichaelO where the MP is picking up it's parameters from for the Tracker.  Maybe it's always going against master for the parameter descriptions even when using the official tracker release.