ArduPilot/ardupilot #4272: Plane: Logging Issues

SlipstreamFPV, 2016-06-05:
#### Issue details
My logs contain random glitches. Almost every value has some spikes that are unaccounted for. For example my 5V line reported a maximum of 461V which obviously isn't right.
[image]

I question if the glitches are read as data by the autopilot as well and used for navigation and other vital functions or if they are just errors in the log. For example in this flight the Baro reported 0m altitude twice.
[image]

Would this also go on to effect the flight control?

Sometimes immediately following a glitch I think I can see bad data being put through on other parts of the navigation system. For example in the attached pictures and log you can see that CTUN.NavRoll usually sits within the constrained limits but during the glitch both NavRoll and Roll peak past 300 and -300.
[image]
[image]
[image]

Other times a system seems to get out of whack for a while like the speed demand and takes some flight mode switching or throttle changes in auto modes to get it to "reset".
[image]

#### Version
3.5.2

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
My Twin Dream

#### Hardware type
Pixhawk, Dual GPS, Dual ACC, Dual Compass, Digital Airspeed Sensor, Mauch Power Module, RMRC 5V BEC, EZUHF

#### Logs
https://drive.google.com/open?id=0B7z1ne03SLaSclVFWkRMZ3lBMGs

magicrub, 2016-06-06:
As the project grows, and log more things.  It's getting to the point that
some sd cards can't write error free.

magicrub, 2017-01-26:
Logging is usually the raw data, but the filtered data is what the control system uses. Both are not always logged. Mission Planner has done a better job at removing massive spikes which are glitches to SDcard writing on slow SD cards. 

Is this issue still a problem? If not then it should be closed.

closed 2017-01-26 by SlipstreamFPV