ArduPilot/ardupilot #4280: Flight Mode Request: Safe Mode

Jman841, 2016-06-06:
#### Issue details
Currently in all vehicles if the EKF blows up or there is high vibrations the attitude controller can have errors and estimates bad enough to cause the vehicle to crash. I had this issue in Arduplane 3.5.3 due to GPS interference where the only mode safe to fly is Manual, however, manual requires a high level of skill to fly and land the plane that many drone users who did not start out in the RC World do not possess, and I have had this issue with Vibration problems in quadcopters and helictopers where there is no safe mode to control the vehicle. 

What I am looking for is a mode that is labeled Safe Mode or Safety that uses nothing but the gyroscope (As it is the least affected by vibrations and interference) to control the Heli/Multi/Plane for some assistance in getting it back to the ground. I understand the precision of this mode will not be nearly as good as if the system is using all the sensors, however, IMO it is much better than loosing attitude control and having the vehicle stall/crash/flip. This could also be used as a failsafe if the EKF detects it is not reliable to maintain attitude control.
#### Version
All current versions

#### Platform
[  ] All
[  ] AntennaTracker
[X ] Copter
[ X] Plane
[  ] Rover

#### Airframe type
All aircraft

#### Hardware type
Pixhawk

#### Logs

magicrub, 2016-06-06:
A gyro-only solution would still pass through the EKF, although we could force DCM-only solution but I'm not sure if want to support on-the-fly like that via mode change. Something to think about.

Don't fret, you're our lucky idea fairy! Keep'm coming! :)

mboland, 2016-06-06:
I must agree that this seems more than a good idea but essential.
Being an RC pilot, manual mode in plane holds no fears, but it is copter that is concerning.
GPS and Compass are our weakest links in the sensor chain and the whole point stabilise was that we could manually the copter with these out of the equation.
I have had many instances where switching to Stabilise was the only way to save the copter, and when you have $10K in the air that is VERY important.
Now I am finding out that if the GPS blows up or the Compass chucks a wobbly then it is all over, the copter goes in no matter what.
This is a deal breaker for me, especially dealing with larger craft.

Most concerning.

Surely there must be a way of having the old Gyro/Accelerometer mode on standby for emergencies.
Initialise DCM at arm and store the values?
Anything would be better than "if GPS goes down so does your craft"

Please?

rmackay9, 2016-06-07:
The described mode is ACRO without the outer angle controller (which helps maintain the current angle after the pilot stops providing stick input).
I'm definitely all for improving the safety of the system but I wonder has anyone here actually experienced an EKF failure that was so serious that the vehicle lost it's attitude estimate but the gyro sensors will still good?  It's extremely rare and I wonder what the chance is that it will happen to a pilot whose got the proposed mode setup and is skilled enough to use it.

mboland, 2016-06-07:
Thanks for chiming in Randy,
I guess our concerns are really based on paranoid fear of the GPS (and compass) being involved in the good old reliable Stab mode most of us have trained ourselves to switch to in any emergency.
I would not like to take control of, and bring back, a large copter from 600m or more away in Acro though.

I am sure most builders have had strange crashes they can't identify, although most likely hardware, and this new path with EKF adds more variables which is making some, including me, a bit nervous.

I must, of course, defer to your vast knowledge of the code and it's abilities, and will always do what I can to help. 
I look forward to trying all this new code and verifying your confidence in it.

Jman841, 2016-06-07:
I saw this issue in Arduplane when a VTX caused major interference with the GPS. All modes, including FBWA, were unsafe to use except for manual. I am not sure if the same effect would happen for the copter attitude control, however, if it did there would be no mode to save it currently with the same GPS Jamming that occured with my setup on Arduplane.

I'm not sure, Randy would know better, is it possible to have these issues in arducopter as well, or is the attitude controller affected differently by the EKF in arducopter than plane?

rmackay9, 2016-06-07:
@Jman841, the EKF is shared between Copter and Plane but the EKF tuning and the code release schedules are different.  So for example, the official version of Copter (3.3.3) is still on the EKF1 while the official version of Plane is on EKF2.
There were definitely issues with the EKF2 uncovered in master (i.e. before official Copter 3.4 beta testing began) and those issues are still in the current Plane release but a patched plane release is going out soon I hear.  So if this problem you saw was with Plane was from a recent release then I think it's a real bug that needed fixing (and was fixed).  For me, that bug alone wouldn't justify creating a new backup flight mode in Copter - better to focus on squashing those kinds bug and ensuring they don't get out.

rmackay9, 2016-06-07:
By the way the pre-EKF estimator, DCM, also used the GPS and we've had similar conversations a couple of years ago to allay similar fears re DCM so it's really not a new behaviour.  Most often people noticed the compensation because the HUD would move around a little while the vehicle was stationary on the ground.  You won't see that now because we put a patch in to turn off the compensation while the vehicle is disarmed.

It's not super obvious but the attitude estimator needs to to compensate for rapid acceleration or deceleration of the vehicle.  If it doesn't then if the vehicle is flown in a circle the accelerometers will see a constant outward pull and this leads to the vehicle leaning into the circle more and more.  With the GPS base compensation we can subtract the acceleration out so the vehicle's attitude stays good.  This is just one obvious example of where it matters, but you can imagine more likely examples including when the vehicle comes to a rapid stop.

In slow moving multicopters doing gentle maneuvers the GPS based compensation isn't super important but it's really important for planes and helicopters.  If someone is flying a very fast multicopter (like Leonard's black football like copter from AVC a few years ago) it also matters.

Anyway, not trying to bury the problem, just saying that it's not actually a new issue and personally I think we should focus on making the EKF robust more than anything else.

Jman841, 2016-06-07:
Thank you for the indepth explanation. For the current plane that I have issues with, would it be safer to use EKF1 than EKF2 while we wait for the new release?

magicrub, 2016-06-07:
Plane release is a couple days away.  Hopefully you can fly out this
weekend.

R-Lefebvre, 2016-06-07:
Randy, to answer your question about whether or not the EKF has ever blown up, and the copter saved in Acro mode, I did it last summer.  The EKF blew up on a test flight of AC3.3RCsomething.  IIRC, it actually happened while I was in Acro.  In this particular case, the AHRS kept switching between EKF and DCM every few seconds.  This led to the copter jerking around 10° every time the AHRS switched, due to Acro's outside attitude observer.

I managed to get it down mostly successfully.  It crashed but suffered no damage.

That turned out to be an EKF bug that was fixed.  So it may seem like a rare case.  But it's not necessarily.

Even with AC3.3 Stable, I can get the EKF to complain quite easily by some maneuvers while sport flying my little red heli.  I fly my heli with the GCS beside me so I can hear all the EKF position and velocity errors as they come in, and temper my flying maneuvers to avoid too many of them happening too fast...  I'm not kidding.

Point being, I'm quite sure I can blow up the EKF quite badly if I wanted it.  In fact, the easiest way to do it, is to do the one maneuver a heli excels at: Rapid yawing.  I would love to get you a log that shows the EKF completely blowing up, while the gyros remain good, but I'm not going to do that because it will cost me a helicopter.  

I would love to do the sort of testing to failure needed to prove this issue. (Or demonstrate the sorts of things that can break the EKF, so Paul can have some data to make EKF more robust).  But I can't because it's going to cost me a helicopter.  Which in itself, is a clear demonstration of the problem.

R-Lefebvre, 2016-06-07:
I do agree with Randy, that the GPS data fusion into the EKF (and the DCM before it) is important.  I still remember the day 3-ish years ago, when I first tried Jon's GPS-DCM fusion code.  I had been doing lots of sporty Stabilize flying with my helicopter, and didn't realize how bad DCM was without GPS.  Nobody had really tried Jon's code before, but that's when the light went on for me.  It was a night and day difference for Stabilize fidelity.  (Similar to the first time I tried Leonard's Earth-Frame-Body-Frame translation code which was also transformative).

However, I've also done lots of Acro flying with the Gyro-Only code.  I developed the error-integrating Acro code that existed in AC3.1 (IIRC)  before Leonard attitude-referenced it for AC3.2.  It was fine.  Honestly I could not detect any difference between the two Acro modes. Both were far better than the pure-gyro non-integrating acro code of AC3.0.

R-Lefebvre, 2016-06-07:
Like Randy said, the GPS fusion is nothing new.  It's been in DCM for like 3 years.  However, I never saw DCM blow up as badly as the EKF can.  Not saying DCM is better.  It is worse at other things.

The main point here, is the need for a back-up piloting mode for copters.  Even if it's only so I can have a fall-back mode so I can torture the EKF to help improve it.  I can't now.

I will say that, I find it troubling that Tridge has created a lot of pass-through code for helicopters, to allow usage of a hobby Flybarless controller for rate control, because he felt it's important to have a backup for when things go wrong in the Pixhawk.  

I don't like the idea of having a separate rate control gyro system operating underneath Arducopter.  IMO, that is a reversion of what I've been working on for 5 years; making Arducopter capable of flying a helicopter all by itself (and which many people are successful with).

But what Tridge is doing has merit for exactly the reason we are talking about here:  He has the ability to land the helicopter after the EKF blows up.  The Pixhawk is certainly capable of doing the exact same thing internally.  This is what we're asking for.

rmackay9, 2016-06-08:
If someone wants to write this acro mode, test it thoroughly, document it, do the user support for it and continue to maintain it going forward then I'll accept it into master.  Perhaps that sounds like a long list of requirements but that's what I'd have to do if I did it myself.

R-Lefebvre, 2016-06-08:
Sure, if it would be included I'll try to find the time to work on it.  Wasn't looking for you to do it.  Shouldn't be that hard to do as the code existed in AC3.1.  Just would need to be updated.