ArduPilot/ardupilot #4281: Replay broken when built with WAF

peterbarker, 2016-06-07:
#### Issue details

Replay has code in it which triggers off APM_BUILD_TYPE.  I'm thinking that's no longer available in waf, so currently Replay tries to write into a directory it has no permission to (/var/APM).

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_Linux/Storage.cpp#L25

What's the waf equivalent of APM_BUILD_TYPE?

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[X] Replay

#### Airframe type
With Replay, you choose!

#### Hardware type
N/A

#### Logs
N/A

OXINARF, 2016-06-07:
APM_BUILD_TYPE does exist in waf. The issue with Replay is that it is using the generic (no specific vehicle) compiled library - https://github.com/ArduPilot/ardupilot/blob/master/Tools/Replay/wscript#L11
What we need to do is compiling the libraries with Replay set as a vehicle - like we do, for example, for AntennaTracker: https://github.com/ArduPilot/ardupilot/blob/master/AntennaTracker/wscript

guludo, 2016-06-30:
Hi @peterbarker ,

I hope #4406  fixes it.

Thanks @OXINARF!

lucasdemarchi, 2016-06-30:
I hope one day these ifdef inside the libraries for the build type are gone.

OXINARF, 2016-06-30:
@lucasdemarchi I think we all do.

guludo, 2016-06-30:
@peterbarker #4406 was applied. If that fixes your problem, we can close this :-)

peterbarker, 2016-06-30:
Thanks all, Replay now runs for me when built with waf!

closed 2016-06-30 by peterbarker