ArduPilot/ardupilot #4282: Terrain data is not being fetched for mission items without GPS

magicrub, 2016-06-07:
Terrain data is not being fetched for mission items without GPS

lvale, 2016-06-07:
Testing: 

Parameters Terrain_Follow=1 Terrain Spacing=10 Terrain Enable=1.

Plugged PixhAwk via USB to Latest MP beta. Draw a mission on MP . Confirmed that on the Mission Planner/SRTM dir some files were downloaded by MP (Zip and then HGT). Uploaded mission to copter but strangely the status screen from the PixHawk keeps showing the following vars all zero: ter_curalt, ter_alt, ter_load, ter_pend, ter_space.
Only thing I must confirm is to have the PixHawk retrieve GPS location (currently indoor without GPS), but I believe that shouldn’t be required to preload the Terrain Data..
If the mission is loaded, shouldn’t the required elevation files for the mission be preloaded, regardles of the current state of the PixHawk being without GPS reception ?

lvale, 2016-06-20:
A mechanism to pre-fetch terrain data to the sdcard could be discussed with the GCS owners

rmackay9, 2019-03-27:
Another bumped into this today.  It's a known issue but just wanted to add a note that it does affect users occasionally.

CodingGhost, 2020-04-12:
I also would love a feature to just mark an erea in MP and upload all that terrain data, it could be so easy!

rmackay9, 2020-04-13:
@CodingGhost, If users have a mission defined though AP will automatically fetch the required terrain data from the ground station so the user shouldn't need to worry.  We actually have very few reports of terrain failsafes which is what happens if the vehicle needs terrain data but doesn't have it.

I think I understand the suggestion but I'm wondering if it's really required.

CodingGhost, 2020-04-13:
Well, from my understanding it also works in fbwb, cruise and rtl. i live in mountain area, and If I could upload terrain data, RTL could be much safer, currently I would set the rtl hight super high, which drains a lot of battery in a emergency situation.
It would be nice for cruising also.
Btw, what spacing do we need between the lines to have full terrain data covering?