ArduPilot/ardupilot #4285: Copter 3.4rc1 add abylity to switch between optical flow or gps setup from aux ch_op switch

roque-canales, 2016-06-07:
Hello,

I just make test of optical flow with EK2, result are pretty good.
Could you please add aibility to switch simply between optcal flow setup and gps?

Add this function waiting ability for gps / flow fusion with EK2.

Also,  there's  already planed to go out with this feature? already available in the px4 stack...

IamPete1, 2021-08-06:
We have a example of how to do this via scripting:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/ahrs-source.lua

closed 2021-08-06 by IamPete1

rmackay9, 2021-08-06:
We should make a simpler script for just GPS+optical flow and we have that on the issues list as well. https://github.com/ArduPilot/ardupilot/issues/15859