ArduPilot/ardupilot #4293: External USB and LED adapter not working

onurboy, 2016-06-08:
#### Issue details
I have been using pixhawk for six months and I am very satisified with it. External usb&led adapter is a good option to download the dflogs from the plane because its not possible to reach the side usb port of pixhawk in the fuselage.

Last week I bought 2 new pixhawk boards and 4 led&usb adapters but I could not make a connection between the boards using external adapters. When I connect it powers the board, pixhawk starts but no light on the external led and pc does not recognize the pixhawk itself.
When I use the default usb port (side) pc recognizes the pixhawk.

First, I thought that one of the boards or adapters was broken but what is the possibility of 2 broken boards or 4 broken usb&led adapters at the same time?

Do you have any idea about what has been changed because I was able to use the led&usb adapter on my previous plane?

Thank you.
Onur.

#### Version
_3.5.3_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Skywalker 1880

#### Hardware type
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-06-08:
I suspect a hardware problem, where did you get the pixhawk from?

onurboy, 2016-06-08:
@magicrub 

Tom,
Thank you for your interest.
I bought it from a chinese vendor. I had bought some items from him before.
Actually the quality of the products look good. But here is the probem...

magicrub, 2016-06-08:
I'm sure it's a quality product and I'm sure they'd be happy to support you if there's a problem with it. Do you have another one to compare to confirm it is not a software problem? otherwise, it's a customer service issue with your vendor

onurboy, 2016-06-08:
Actually currently I have 2 pixhawk boards and 4 led&usb adapters. I've tried it with all. No way for both pixhawk boards and 4 usb&led adapters. I will get in touch with the vendor but I am not sure which components are broken, pixhawk boards or usb&led adapters?

rmackay9, 2016-06-09:
maybe check that there's an SD card inserted?  I'm closing this 'cuz it's really not a software issue, at best it's a support question which should probably be raised in the forums instead of on github.
http://discuss.ardupilot.org/

closed 2016-06-09 by rmackay9