ArduPilot/ardupilot #4294: Feature Request: Tricopter Dynamic Yaw Input

nmrdirtman, 2016-06-09:
#### Issue details
_Please describe the problem, or desired feature_
Would like to make a request to consider adding dynamic yaw input from the yaw servo in a tricopter.  Triflight ( https://github.com/lkaino/Triflight/releases ) uses this feed back data.  Also RCExplorer.de sells a modified Blue Bird 210DMS servo with a feedback lead.  http://rcexplorer.se/product/blue-bird-bms-210dmh-servo/  It uses a magnetic induction rotary encoder to know the rotor angle which can be used in calculations.  This seems like a good idea.  Please forgive me if its stupid or already address.  I am willing to test.  
#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

ChrisKrieg, 2017-10-18:
This is quite an interesting enhancement. Has there been any development on implementing this sensor feedback?

nmrdirtman, 2017-10-18:
No a word.  I am not a good coder but I bet if you look at at the triflight source code and port the control algorithm to Ardupilot it could work.  Just beyond my abilities.

IamPete1, 2019-11-02:
@nmrdirtman @ChrisKrieg I have done a fix for this issue, It would be great if you could help me test it

doceave, 2020-07-02:
Hi there IamPete1... I have a tricopter under construction --- the same as described above by RCExplorer.se using but using a KakuteF7 AIO FC

I would be very happy to test the performance of the servo feedback.

Can you please advise how I can get hold of the 

Please advise how I can get a firmware version with this feature for the F7 AIO

IamPete1, 2020-07-02:
@doceave [one build for Kakute F7](https://drive.google.com/file/d/1NqXTCk-rAsDAe64zqJ4P3oSIt--SilkT/view?usp=sharing)  

This is built on top of master, so it may be a little different if you have been flying stable. Set YAW_SV_PIN to the analog pin the feed back from the servo is connected to, RSSI pin is often used. Although I'm not sure what pin number that would be. Then set YAW_SV_SPEED to -1 to trigger a auto calibration. Have a look in the messages tab to see how it got on. You should then be good to fly.

doceave, 2020-07-02:
Many thanks indeed Peter!

I am happy to repeat the tri configuration. Perhaps I will make some
baseline recordings before implementation of tail servo feedback....

Please just confirm for me the process (I am but a simple end user of this
amazing software):
> I update the firmware on the F7 using the .apj linked
> I adjust the parameters YAW_SV_PIN and YAW_SV_SPEED using MissionPlanner
as you described
> That's it!?

How is the file you link different from the latest .hex provided for the
Kakute F7?

Thanks

IamPete1, 2020-07-02:
@doceave Yep that's it, if you leave YAW_SV_SPEED at 0 then its just a old fashioned tricopter. The new params have a MOT_ prefix, so the full thing would be MOT_YAW_SV_SPEED MOT_YAW_SV_PIN, there are some voltage ones too, but they should be auto populated by the calibration. 

The .apj is exactly the same code as the .hex just packaged differently. Once you have uploaded a xxxx_with_bl.hex using betaflight configorator you then have the AP bootloader and can upload .apj's with Mission Planner.

doceave, 2020-07-02:
I think I'm almost good to go!

Any idea how I would find out the pin number used for RSSI by the FC?

The Kakute F7 has an STM32F745 at its core --- this has 3ADCs; but without
a schematic of the FC it would still be tricky to determine which GPIO is
connected to the RSSI pin on the board... ;)

Perhaps somone on this forum knows?

IamPete1, 2020-07-02:
@doceave Yeah worth a search on the forum, shouldn't take too long to just try a few, the param description gives the common ones. If you can work it out we should add it to the documentation, If not I have one here somewhere I can see If I can work it out.

doceave, 2020-07-02:
I endeavor to figure it out! :) There are only a handful of possibilities
according to the STM32F745 datasheet.

Will revert with an answer soon hopefully.

doceave, 2020-07-16:
> @doceave Yeah worth a search on the forum, shouldn't take too long to just try a few, the RSSI_PIN param description gives the common ones. If you can work it out we should add it to the documentation, If not I have one here somewhere I can see If I can work it out.

I have completed my tricopter build and tuned it up to the best of my ability --- slight low frequency tail wobble but tomorrow that I hope to address by tubing Yaw kP/I/D in flight... 

Sadly I have not managed to figure out the pin number associated with the RSSI pad on my FC, a Kakute F7 AIO v1.2 

@IamPete1 --- Would you please call on your knowledgeable contacts to help me out re figuring out which pin to set as YAW_SV_PIN?

I will then be able to make good comparison of flight characteristics achievable with and without feedback from the tail servo :)

IamPete1, 2020-07-16:
@doceave I had a trawl through the code, its pin 15

doceave, 2021-11-28:
@IamPete1 ---- More than a year and some rebuilds later... Tricopter files nicely using the latest stable firmware without any servo feedback. There is, however, some tail wobble that is only bothersome when flying FPV.

Would you kindly again provide me with link to firmware for the Kakute F7 AIO with servo feedback? This time I hope to actually implement it properly.

Thanks.

marcdornan, 2021-12-15:
@IamPete1 Finally I built my mini tri! can probably just compile from a clone of your fork.However, I am not having much joy cherry picking that commit into my refeshed my fork of master. Here is what I get. Any ideas what the issue may be?

$ git cherry-pick 1c9c179130d4ef5ea777b01f6c87dd6720b4a41d
Auto-merging libraries/AP_Motors/AP_MotorsTri.h
CONFLICT (content): Merge conflict in libraries/AP_Motors/AP_MotorsTri.h
Auto-merging libraries/AP_Motors/AP_MotorsTri.cpp
CONFLICT (content): Merge conflict in libraries/AP_Motors/AP_MotorsTri.cpp
error: could not apply 1c9c179130... AP_Motors: tricopter: estimate and measure yaw servo position

IamPete1, 2021-12-16:
@marcdornan I have rebased the branch. 

@doceave This is a Kakute F7 build https://drive.google.com/file/d/1N5RKUplQBa4Y_vTq-2GZhyARHIS95i8-/view?usp=sharing

doceave, 2023-05-13:
@IamPete1 --- My tricopter has now returned for a 3rd time! This time around I endeavor to do some tail servo testing before the vehicle attempts an RTL through a mountain!

Would you be willing to post a fresh link to a tricopter build with the servo angle feedback? Thanks.