ArduPilot/ardupilot #4297: Copter/Plane: Special beep tone for GPS 3D lock and safety arming without GPS lock?

GrahamDyer, 2016-06-09:
#### Issue details
There doesn't seem to be a specific audio tone for when the 3D GPS lock occurs after booting. This would an incredibly useful enhancement especially on different platforms like the Pixracer and AUAV-X1 and X2. Currently there is no way to determine when these FC's have GPS lock as the LED's don't show it. Just from a safety point of view it would be a huge help to ensure that one would wait for this tone before taking off to ensure that 3D GPS lock has been attained.

Also a parameter to disallow arming in any mode (not just nav modes) without GPS lock would be a very important safety addition. Most users always need GPS lock for safe flying, those users who don't need GPS lock (indoors, racing, etc) can then enable arming without GPS lock.

#### Version
All versions

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Plane and copter
#### Hardware type
AUAV-X2

#### Logs
N/A

rmackay9, 2016-06-09:
I agree especially about the need for an optional pre-arm check to ensure we only arm once we have 3D lock.  It's been discussed before I'm certain.

GrahamDyer, 2017-07-19:
Any progress with these: 
1) a tone to indicate 100% ready to go, GPS 3D lock obtained? 
2) a "can't takeoff without GPS 3D lock" parameter?

kd0aij, 2017-07-28:
My AUAV-X2 running ArduPlane/EKF3 beeps when the EKF starts using GPS. Prior to that it won't arm in FBWA or QSTABILIZE modes. I don't know if Copter behaves the same.

Pedals2Paddles, 2017-07-28:
Remember it isn't 3D GPS lock you need to know about. It's the EKF deciding the GPS position is usable.

peterbarker, 2024-07-31:
We have an option to enforce must-have-position-before-arming now.

We also have both "happy sounds" for ready-to-arm and sad-sounds for no-longer-ready-to-arm.

Closing this one.

closed 2024-07-31 by peterbarker