ArduPilot/ardupilot #4298: Tricopter V3.4-rc1: uncommanded yaw when flying backwards

GrahamDyer, 2016-06-09:
#### Issue details
When flying backwards my aircraft exhibits an uncommanded yaw behavior. This was not apparent in v3.3.

[image]
.

#### Version
AC V3.4-rc1

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Tricopter

#### Hardware type
AUAV-X2

#### Logs
https://www.dropbox.com/s/6ormmj1elpjdne5/95%2008-Jun-16%2016-42-56.bin?dl=0

R-Lefebvre, 2016-06-11:
I can't download your logs form Dropbox to check.  This could be a yaw "blowout".  Basically if the yaw control is not capable of controlling the yaw due to backward flight, it blows around.  Very common on helicopters.  Not surprised to see a tricopter do it too.

Can you plot ATT.YawDes vs. ATT.Yaw?

GrahamDyer, 2016-06-11:
Hi Rob, try this link: http://filebin.ca/2kJ1J4LGer7U/9508-Jun-1616-42-56.bin

Here is ATT.YawDes vs. ATT.Yaw

[image]

rmackay9, 2016-06-13:
Really we should move this discussion to the discourse servers.  http://discuss.ardupilot.org/c/arducopter/copter34

Perhaps as Rob says, it looks like the target yaw is being dragged around by the actual yaw.  I think that means that the yaw control of the vehicle (whether because of tuning or because of some physical issue) is not capable of maintaining the yaw.

OXINARF, 2016-11-02:
@GrahamDyer Have you tested this with the 3.4.1 stable release? Is the issue still present?

OXINARF, 2016-12-17:
No answer for one month and a half, closing.

closed 2016-12-17 by OXINARF

GrahamDyer, 2016-12-20:
Sorry, I had replied but not in the issue: http://discuss.ardupilot.org/t/ac3-4-rc1-tricopter-test-flight-a-couple-problems-noted/9405/8

"I can confirm that the uncommanded yaw is no longer present. The tricopter maintains it's heading while flying backwards."

OXINARF, 2016-12-20:
No problem, great to hear it's working. Thanks for testing!