ArduPilot/ardupilot #4302: Camera shutter doesn't work in V3.4-rc1

buckker, 2016-06-11:
The camera shutter doesn't work in arducopter 3.4-rc1. I have no problems with arducopter V3.3.3 and Mission Planner 1.3.36. I only tested it with the Mission Planner function "Camera Trigger NOW"

cczeets, 2016-06-14:
I'm not having any troubles with it.  I've ran several missions with 3.4-rc1 and also used the "Camera Trigger NOW".  They have both worked great for me!  I've tested it on 3 different camera and 3 different triggers as well.  I've tested a Sony NEX-3N, NEX-5N, A7 with a Seagull, Flytron, and hard soldered triggers.

Best,
Steve

buckker, 2016-06-14:
Hi Steve

Ok, that's strange. Which output do you use for the camera trigger? I use the pin no. 2 (RC10) Can you upload your settings or maybe do some print screens from the gimbal page in Mission Planner?

Does the camera trigger also work on the ground when the copter is disarmed? 

Thanks a lot

Michael

cczeets, 2016-06-14:
Michael,

I've found that using RC7 is the most reliable way to trigger a camera, not sure why.  I previously used RC9 or RC10 (quite a few revisions ago), but had issues with intermittent triggering.
The copter must be armed to trigger the camera for me on RC7.  Yes, it even works on the ground.  But, I must press the arm button for it to trigger.

Best,
Steve

buckker, 2016-06-14:
You're my hero :-) It really works when the copter is armed. That's new compared to V3.3.3. 

Thank's a lot Steve

@rmackay9 What's the idea behind this?

cczeets, 2016-06-14:
I noticed that was new as well, but didn't think anything of it...
Glad you got it working!

OXINARF, 2016-11-02:
This was changed for the 3.4 release. Any non-motor output is now enabled by default.

closed 2016-11-02 by OXINARF